RehabMan/OS-X-Voodoo-PS2-Controller #50: Sometimes keyboard not working after sleep

qiuyi21, 2015-05-04:
I'am using OS X 10.10.3 with Acer Aspire 4741G, and the output with command "kextstat | grep -i ps2" as follows:
   34    2 0xffffff7f80ee0000 0xb000     0xb000     org.rehabman.voodoo.driver.PS2Controller (1.8.14) <11 7 5 4 3 1>
   43    0 0xffffff7f80f03000 0x6000     0x6000     org.rehabman.voodoo.driver.PS2Keyboard (1.8.14) <35 34 7 6 5 4 3 1>
   45    0 0xffffff7f80efd000 0x4000     0x4000     org.rehabman.voodoo.driver.PS2Mouse (1.8.14) <35 34 5 4 3>

Sometimes keyboard not working after sleep, I must reboot system to resolve it. Why and How to avoid this?

RehabMan, 2015-05-23:
Use the debug version to gather more information.

RehabMan, 2015-08-14:
No response.  Closing.

closed 2015-08-14 by RehabMan