RehabMan/OS-X-Voodoo-PS2-Controller #64: ALPS v6 drag delay problem

florentino97425, 2015-08-07:
Hi, 
I use voodoo radut alps v6 kext,
I have a problem with double click to drag.
When I try to drag a finder window or everything else with double click with my trackpad, it don't drag immediately, it take 2-3 sec to the double click, drag activate. 
Can someone help me, please.
Thank.

https://github.com/radut/OS-X-Voodoo-PS2-Controller/tree/alps_v6

RehabMan, 2015-08-14:
I'm not supporting anything but my version.  Report your problem to the author.

closed 2015-08-14 by RehabMan

florentino97425, 2016-06-04:
Thanks.
Fixed, for who have same issue try to edit VoodooPS2Trackpad.kext info.plist.
For me, I fix it with.

	<key>Default</key>
	<dict>
		<key>ClickPadClickTime</key>
		<integer>300000000</integer>
		<key>ClickPadTrackBoth</key>
		<true/>
		<key>DisableDevice</key>
		<false/>
		<key>DoubleTapThresholdX</key>
		<integer>100</integer>
		<key>DoubleTapThresholdY</key>
		<integer>100</integer>
		<key>DragExitDelayTime</key>
		<integer>1000000000</integer>
		<key>DragLockTempMask</key>
		<integer>1048592</integer>
		<key>ImmediateClick</key>
		<false/>
		<key>MaxDragTime</key>
		<integer>180000000</integer>
	</dict>