RehabMan/OS-X-Voodoo-PS2-Controller #66: Can I use two finger gestures ?

MichelBahl, 2015-09-05:
I am using an Lenovo Y70.
The trackpad is working except multitouch gestures,
what do I need to do to make it work ?

RehabMan, 2015-09-15:
Two finger gestures:
- two fingers scroll
- two finger tap for right click
- two finger double tap for double right click

It is not true multitouch, but rather simulation.

closed 2015-09-15 by RehabMan

MichelBahl, 2015-09-15:
Unfornately I can't get it working.
If I install voodoo-ps2 the basics are working the touchpad behaves like a mice.

It's really annoying when I want to scroll, I always need to hit the small scroll
bar.
I downloaded the source code for setup remote kext debugging,
but it seems since OSX 10.9 a developer certificate is required.

Can you give me tip, maybe I did something wrong at installation.

RehabMan, 2015-09-15:
Perhaps you don't have a Synaptics trackpad.