RehabMan/OS-X-Voodoo-PS2-Controller #69: VoodooPS2Daemon does not work on 10.11

RehabMan, 2015-10-03:
And... it looks like they broke CF based ioreg watching/iteration pretty badly.

May need to find a way to do this from the kext itself.

philippetev, 2015-10-03:
So, it's not just my imagination...

RehabMan, 2015-10-04:
I think this will be useful...

/*! @function addMatchingNotification
    @abstract Adds a persistant notification handler to be notified of IOService events.
    @discussion IOService will deliver notifications of changes in state of an IOService object to registered clients. The type of notification is specified by a symbol, for example <code>gIOMatchedNotification</code> or <code>gIOTerminatedNotification</code>, and notifications will only include IOService objects that match the supplied matching dictionary. Notifications are ordered by a priority set with <code>addNotification</code>. When the notification is installed, its handler will be called with each of any currently existing IOService objects that are in the correct state (eg. registered) and match the supplied matching dictionary, avoiding races between finding preexisting and new IOService events. The notification request is identified by an instance of an IONotifier object, through which it can be enabled, disabled, or removed. <code>addMatchingNotification</code> does not consume a reference on the matching dictionary when the notification is removed, unlike addNotification.
    @param type An OSSymbol identifying the type of notification and IOService state:
<br>	<code>gIOPublishNotification</code> Delivered when an IOService object is registered.
<br>	<code>gIOFirstPublishNotification</code> Delivered when an IOService object is registered, but only once per IOService instance. Some IOService objects may be reregistered when their state is changed.
<br>	<code>gIOMatchedNotification</code> Delivered when an IOService object has been matched with all client drivers, and they have been probed and started.
<br>	<code>gIOFirstMatchNotification</code> Delivered when an IOService object has been matched with all client drivers, but only once per IOService instance. Some IOService objects may be reregistered when their state is changed.
<br>	<code>gIOTerminatedNotification</code> Delivered after an IOService object has been terminated, during its finalize stage.
    @param matching A matching dictionary to restrict notifications to only matching IOService objects. The dictionary is retained while the notification is installed. (Differs from addNotification).
    @param handler A C function callback to deliver notifications.
    @param target An instance reference for the callback's use.
    @param ref A reference constant for the callback's use.
    @param priority A constant ordering all notifications of a each type.
    @result An instance of an IONotifier object that can be used to control or destroy the notification request. */

    static IONotifier * addMatchingNotification(
                            const OSSymbol * type, OSDictionary * matching,
                            IOServiceMatchingNotificationHandler handler,
                            void * target, void * ref = 0,
                            SInt32 priority = 0 );

Polishapple, 2015-10-10:
What that daemon does?

RehabMan, 2015-10-10:
> http://www.tonymacx86.com/yosemite-laptop-support/164990-faq-read-first-laptop-frequent-questions.html

It is described in the README/changelog.

RehabMan, 2015-10-16:
This will be fixed in the 1.8.16 build... but I have a few other things I want to fix.

I fixed it in VoodooPS2Daemon, so the daemon is still required.

RehabMan, 2015-10-17:
Fixed in 2015-10-16 build 1.8.16.

closed 2015-10-17 by RehabMan