RehabMan/OS-X-Voodoo-PS2-Controller #75: Scrolling with TrackPoint registers a middle click (T420)

SilverEzhik, 2016-03-25:
**Expected behavior:** A middle click is only registered if the middle button is pressed on its own. If the TrackPoint nub is touched and scrolling is initiated, a middle click is not registered. 

This is the way TrackPoint scrolling behaves on Linux and on Windows with TPMiddle.

**Actual behavior:** A middle click is registered even if scrolling was initiated. This results in weird behavior with some apps:

* Sublime Text - [it interferes with its middle click selection feature](https://gfycat.com/ChiefPerkyArrowana)
* iTunes - Choppy scrolling, inability to scroll in playlist view
* Chrome - inability to scroll over images that have interaction if you click on them.

musashiz, 2016-04-27:
Also having this issue. Id be interested in knowing how to disable middle click as I only use middle button for scrolling. Or maybe change some value? 

I changed FakeMiddleButton to false in VoodooPS2Mouse and it had no effect from what I could tell.

jakibaki, 2016-06-15:
Also have this issue on my thinkpad x220 :(

RehabMan, 2016-08-22:
Source code is available.  Suggest you debug it to determine the source of the problem.  I don't have your hardware.  Good luck.  If you find a solution (eg. changes the the code), you can submit a pull request.

closed 2016-08-22 by RehabMan