RehabMan/OS-X-Voodoo-PS2-Controller #78: Cube i7 problems

byzyn4ik, 2016-06-22:
I don't found a key in plist to disable very bad feature of driver with my trackpad. My trackpad was recognized as mouse , all works but when i am moving finger from top to bottom they sending cmd+z combination. When moving from left to right  sending cmd+backspace  , from right to left cmd+c , from bottom to top - nothing all works like i need. I have small knowledge of c++ but big knowledge of obj-c and c , so if you can give me advice where i need to disable this , i will try.

RehabMan, 2016-08-22:
Define 'they' in 'they are sending'.

byzyn4ik, 2016-08-23:
May be driver sends cmd-z when trackpad moving from absolute top to bottom (if a moving trackpad from center i have no problems).

RehabMan, 2016-08-24:
My code doesn't.  It is possible (but a bit strange) that your trackpad firmware does it though.

At any rate, not related to this project.

closed 2016-08-24 by RehabMan