RehabMan/OS-X-Voodoo-PS2-Controller #80: Keyboard works, touchpad gives error (PS2)

api984, 2016-07-09:
Hello,

Keyboard was loaded. 

Touchpad gives error and does not work.

Jul  9 17:58:46 localhost kernel[0]: ApplePS2Controller: Timed out on mouse input stream.
Jul  9 17:58:46 localhost kernel[0]: ApplePS2Controller: Timed out on mouse input stream.
Jul  9 17:58:46 localhost kernel[0]: ApplePS2Controller: Timed out on mouse input stream.
Jul  9 17:58:46 localhost kernel[0]: ApplePS2Controller: Timed out on mouse input stream.

kexts:
sh-3.2# ls | grep -i ps2
VoodooPS2Controller.kext
VoodooPS2Keyboard.kext
VoodooPS2Trackpad.kext

No apple PS2 kexts installes. OSX: El Capitan, 10.11

Any advice?

api984, 2016-07-09:
I havent been able yet to detect what model ore make is it. Asux X455LA laptop. Asus site gives its own Touchpad name. Linux did not say anything useful yet about make and model.

Still trying to detect it. If its ELAN, Synaptics or other.

RehabMan, 2016-08-22:
Look in ioreg.  It might not even be PS2.

RehabMan, 2016-09-23:
Closed due to lack of response.

closed 2016-09-23 by RehabMan