RehabMan/OS-X-Voodoo-PS2-Controller #81: Caps lock needs double click

Fgsaramago, 2016-07-21:
I need to click the caps lock key two times for it to change between lowercase and uppercase. My keyboard is a SteelSeries, connected through ps2, of course. This didn't happen with the voodoops2 kext I had in Yosemite

RehabMan, 2016-08-22:
What version of OS X/macOS?

Fgsaramago, 2016-08-22:
Running Sierra public beta, now updated to beta 5 and the issue still persists. I also tried with another ps2 keyboard, a very old unbranded one I have and it has the exact same issue. It always needs two clicks on the Caps Lock key to do something. It doesnt need to be two fast clicks like double clicking on a mouse, it just needs to be clicked two times

RehabMan, 2016-08-22:
Known problem with 10.12.

transtone, 2016-09-22:
when will fix it?

RehabMan, 2016-09-22:
It is not nice to ask developers 'when'.

transtone, 2016-09-24:
sorry, Is there any possible plan to fix it ? 
How Can I help this, not only bothring a developer?

RehabMan, 2016-09-24:
I'll look at it when I have time.  Currently, there are many more issues with 10.12, which quite frankly prevent its use (I'm not even installing it on my real Mac).

So... low priority.

MatthewHKnight, 2016-10-22:
I don't believe this is limited to ps/2 I'm using a USB keyboard just a random keyboard I thought I was having issues with my old keyboard so I bought a new keyboard this issue is happening with the new keyboard I think I'm likely gonna reinstall OSX El Captain.
All my other keys work just fine just not my caps lock it takes a few presses on the caps to get it to register the key was pressed down.
Funny thing is everything worked fine on OSX El Captain few days after upgrading to macOS Sierra my caps lock started acting up so as I said I went out bought a new keyboard and still happens finally I google'd caps lock macOS Sierra and see others are having the same issue.
reminds me of Windows upgrades all over again.

usr-sse2, 2016-11-09:
Latest commit in #90 contains a workaround to Caps Lock issue on Sierra, not tested on older systems.

transtone, 2016-11-13:
https://github.com/tluck/OS-X-Voodoo-PS2-Controller 

Does this fixed the Caps Lock bug?

tluck, 2016-11-14:
@transtone - Yes - my branch includes all these fixes/mods for Lenovo T420 and T460.
cap locks fix
Pref Pane 
Smooth scrolling
and i just posted a fix for TrackPoint scrolling - so scrolling in Safari does not open a bunch of links for example.

MatthewHKnight, 2016-11-14:
Eh I assume this is for Laptops my issue is for a Desktop Hackintosh. Also I'm using a USB Keyboard now the LUX K70

transtone, 2016-11-15:
@tluck  I can't open issue in your copy of project, I ask here:
the trackpoint can't work in my laptop(x1carbon 2015), and the middle key work can't sticky on scroll a page.
could you please merge this project on it's trackpoint fix? 
https://github.com/jcsnider/OS-X-Voodoo-PS2-Controller

tluck, 2016-11-15:
@transtone - i will take another look at jcsnider code - it was based on a pretty old version. i have T460 and may be similar as X1. i just got TrackPoint scroll working for the older T420 and realized it is not the same as T460.

usr-sse2, 2016-11-15:
@MatthewHKnight, it's strange that you have this issue on a USB keyboard, on my keyboard caps lock works OK. But this driver can't do anything with USB, it's only for PS/2.

tluck, 2016-11-16:
@transtone - i implemented a mod to make TrackPoint and scrolling work better - if MiddleMouseScroll is true - then middle button is not issued to do a cut/paste. Scrolling works better - hold middle button down and move nob. see here for compiled kext http://www.insanelymac.com/forum/topic/315451-guide-t460-with-clover

transtone, 2016-11-16:
I test the  `T460_UEFI-Clover-b3923_10.11_10.12-2016-11-15.zip`, I can't use my TrackPoint any more.

tluck, 2016-11-16:
@transtone - hmm. what OS ver are you on? does the original version of 1.8.24 work at all on X1? Meaning does the nob move the mouse? my mods to 1.8.24 are really about button changes for this type of ClickPad on T440/T450/T460.  Does the TrackPad (ClickPad) left/right button click work? Do the real buttons do anything? Note: i use an SSDT to define the PS2K device and define the keyboard properties as LENOVO and T460 - which overrides some defaults. But strange if this kext does nothing -  since the jcsnider kext mostly worked on my T460 with El Cap. However, that kext was was based on old code and there was minor logic issues around dragging (with buttons) that  i believe to be fixed.

RehabMan, 2016-12-08:
I implemented a work around in v1.8.25.

closed 2016-12-08 by RehabMan