RehabMan/OS-X-Voodoo-PS2-Controller #82: Lenovo T460 TrackPoint Buttons are all right click?

tluck, 2016-07-22:
the Lenovo T460 has a Synaptics trackpad with "buttons" by clicking lower left/corner - these buttons work as expected when Clicked. However, the TrackPoint buttons above the trackpad (3 - left middle right) all act as if they are right click. what can I send or check to fix this? I'm am 1.8.24 VoodooPS2.  if i switch the mouse primary left/right they all switch to left but i am not lefty! so looking for normal left/right click on these real buttons. thanks.

it has touch screen. mouse moves around. which is cool. but was wondering if there is a way to make a click on the screen

tluck, 2016-08-21:
fixed in a new branch -- added code to handle this type of clickPad buttons.

closed 2016-08-21 by tluck

effeci, 2019-02-21:
Sorry to post on this old issue... I have the same problem with last update. How it is meant to be set the correct way?