RehabMan/OS-X-Voodoo-PS2-Controller #83: ASUS G501 Touchpad loads as Mouse

jagiella, 2016-08-13:
After installing the VoodooPS2Controller.kext as instructed my keyboard and touchpad work well. The only issue is, that I don't have multitouch gestures as the driver is supposed to offer and the touchpad is capable of doing (in Windows it works). After checking which kext are loaded I saw that the VoodooPS2Mouse is loaded instead of the VoodooPS2Trackpad.

Any idea why that is the case, and how I could force to load the trackpad instead?

siddx, 2016-08-15:
Try bpedman_merge branch.

jagiella, 2016-08-15:
I just tried to compile it, but didn't succeed. Is there a repository with the binaries for this branch?

jagiella, 2016-08-15:
Okay I got it compiled, but ended up in a kernel panic at boot time. Maybe because I'm using the new XCode 8 beta instead of XCode 6.

RehabMan, 2016-08-22:
Your trackpad is probably ELAN.  Not supported by this (trackpad) driver.

closed 2016-08-22 by RehabMan