RehabMan/OS-X-Voodoo-PS2-Controller #85: A lot of problem at Sierra

vaslobas, 2016-09-05:
Hello.

I recently update my hack to 10.12 and i got this:
1. Lost function fey functionality. Built-in volumes/brightness control now doesnt works.
2. Touchpad doesnt work properly and touchpad's prefpane in system preference doesn't open.
[image]

Before update to 10.12 all was woks like a charm.

Polishapple, 2016-09-10:
We're on same boat - same thing happend

ThisIsQasim, 2016-09-11:
Yup. I can confirm that happened to me as well.

devbau, 2016-09-22:
Download the patched macOS Sierra Trackpad prefPane*, unzip and move to System/Library/PreferencePanes
Note that this patched Trackpad prefPane will need to be opened once after each restart to make its settings active again. We’re working on a better solution.

* http://www.mediafire.com/download/8r6ew57721w5r82/macOS_Sierra_patched_Trackpad_prefPane.zip
(Source: http://x220.mcdonnelltech.com/thinkpad-x220-macos-sierra-installation/)

RehabMan, 2016-09-23:
Guys... These problems are all well known.

Polishapple, 2016-09-24:
So... We shouldn't post issues?

RehabMan, 2016-09-25:
Obviously, the issue is posted.  But it is not a problem with this kext.  
You can set your needed settings with ioio from a LauchDaemon.

TheCuriousHacker, 2016-09-26:
CAPSLOCK does not work properly..

RehabMan, 2016-09-27:
The capslock problem is well known.   No solution at this time.

TheCuriousHacker, 2016-09-29:
> Obviously, the issue is posted. But it is not a problem with this kext.
> 
> You can set your needed settings with ioio from a LauchDaemon.

Please can you give link to the instructions?

RehabMan, 2016-09-29:
ioio is available from my github.
You can read about LaunchDaemon setup on the internet.

L1v1ng0nThe3dge, 2016-10-18:
On Lenovo Thinkpad X220, volume keys don't work. Same problem with the trackpad. Just reporting findings. Upgraded to Sierra yesterday.

midi1996, 2016-10-29:
Apparently 10.12.1 fixes the Volume and Brightness issues, however CAPSLOCK issues is still present as well as the Touchpad prefpan

usr-sse2, 2016-10-30:
#89 resolves the issue with Touchpad prefpane.

RehabMan, 2016-12-07:
I will be updating the kext soon with some of usr-sse2 changes and my own changes for brightness.

closed 2016-12-07 by RehabMan