RehabMan/OS-X-Voodoo-PS2-Controller #87: Reset & Laptop restart , EC reset doesn't help

indiandragon, 2016-09-27:
Hi RehabMan,

I have been facing the Reset issue on my HP DV6 6154TX using this kext from the beginning. EC reset doesn't help in any ways as I've tried multiple times. One clue I could give is, this often happens when **backspace continuously** on the keyboard . Not exactly sure on how keyboard interferes with touchpad driver, but this issue doesn't occur when I'm using trackpad exclusively for some activity.

This issue is seriously affecting my productivity, any help to resolve this would be much appreciated.

```
Sep 27 15:48:07 --- last message repeated 2 times ---
Sep 27 15:48:07  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (3e) from PS/2 controller
Sep 27 15:48:07  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (00) from PS/2 controller
Sep 27 15:48:07  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (00) from PS/2 controller
Sep 27 15:48:07  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:07  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (41) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (31) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (15) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (10) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (47) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (54) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (00) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (10) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (4b) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (10) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (43) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (4c) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (15) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (10) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (4b) from PS/2 controller
Sep 27 15:48:08  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (64) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (00) from PS/2 controller
Sep 27 15:48:09 --- last message repeated 1 time ---
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (00) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (00) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (59) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (d1) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (00) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (07) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (69) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte3 data (07) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (e0) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (68) from PS/2 controller
Sep 27 15:48:09  kernel[0]: ApplePS2SynapticsTouchPad: Unexpected byte0 data (ab) from PS/2 controller
```

RehabMan, 2016-10-05:
You probably have not installed correctly.  Use the forums.

closed 2016-10-05 by RehabMan

indiandragon, 2016-10-06:
@RehabMan, I'm confident that I followed your guideliness correctly. I have installed your kext several times on different computers. Why is this not related to https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/issues/10 .

RehabMan, 2016-10-06:
Use the forums.