RehabMan/OS-X-Voodoo-PS2-Controller #91: kext crashed every time I shutdown the system 10.12

ensorrow, 2016-11-11:
I'm not quite sure whether the problem is caused by the kext or not, but every time I turn on my computer, it will report a unexpected shutdown cause and present a log like this

```
*** Panic Report ***
panic(cpu 1 caller 0xffffff80086fdd68): "Attempting to lookup/free the middle of a kalloc'ed element! (map: 0xffffff800d1796e8, addr: 0xffffff80a2373da8, entry: 0xffffff8014d05140)\n"@/Library/Caches/com.apple.xbs/Sources/xnu/xnu-3789.21.4/osfmk/kern/kalloc.c:460
Backtrace (CPU 1), Frame : Return Address
0xffffff80a2373c10 : 0xffffff80086f368c 
0xffffff80a2373c90 : 0xffffff80086fdd68 
0xffffff80a2373cd0 : 0xffffff80086fdc70 
0xffffff80a2373cf0 : 0xffffff8008c5815e 
0xffffff80a2373d10 : 0xffffff8008cbfada 
0xffffff80a2373d80 : 0xffffff7f8b1b3c81 
0xffffff80a2373e00 : 0xffffff7f8b1b26cf 
0xffffff80a2373e40 : 0xffffff8008c997ed 
0xffffff80a2373e80 : 0xffffff8008c9953e 
0xffffff80a2373f10 : 0xffffff8008c98af6 
0xffffff80a2373f60 : 0xffffff8008c9a50e 
0xffffff80a2373fb0 : 0xffffff80086a2af7 
      Kernel Extensions in backtrace:
         org.rehabman.voodoo.driver.PS2Keyboard(1.8.12)[A7BBC8BC-C247-3495-B22C-2F51B3D6FA70]@0xffffff7f8b1b0000->0xffffff7f8b1b6fff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[1A48D920-280E-36FA-8D48-49B79A5656E6]@0xffffff7f89495000
            dependency: com.apple.iokit.IOReportFamily(31)[B14DC3D3-7250-3DA3-BF50-C666EBEDAF4C]@0xffffff7f8987a000
            dependency: org.rehabman.voodoo.driver.PS2Controller(1.8.12)[06BBE3FA-265B-384C-A137-C48C6686510D]@0xffffff7f8b1a5000

BSD process name corresponding to current thread: kernel_task
Boot args: kext-dev-mode=1 dart=0 nv_disable=1 -v  debug=0x100 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 16.1.0: Wed Oct 19 20:31:56 PDT 2016; root:xnu-3789.21.4~4/RELEASE_X86_64
Kernel UUID: 75CA1C4D-7BF4-321B-B544-D8F1B6D60EF8
Kernel slide:     0x0000000008400000
Kernel text base: 0xffffff8008600000
__HIB  text base: 0xffffff8008500000
System model name: MacBookPro11,2 (Mac-FC02E91DDD3FA6A4)

System uptime in nanoseconds: 1416465174
last loaded kext at 1208611959: org.rehabman.voodoo.driver.PS2Keyboard	1.8.12 (addr 0xffffff7f8b1b0000, size 28672)
loaded kexts:
org.rehabman.voodoo.driver.PS2Keyboard	1.8.12
org.rehabman.voodoo.driver.PS2Controller	1.8.12
org.netkas.driver.FakeSMC	1723
org.tgwbd.driver.NullCPUPowerManagement	1.0.0d2
com.apple.iokit.IOAHCIBlockStorage	295.20.1
com.apple.driver.AppleAHCIPort	326
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleACPIButtons	5.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleACPIEC	5.0
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	172
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleXsanScheme	3
com.apple.iokit.IOAHCISerialATAPI	266
com.apple.iokit.IOSCSIArchitectureModelFamily	394
com.apple.driver.usb.AppleUSBXHCIPCI	1.1
com.apple.driver.usb.AppleUSBXHCI	1.1
com.apple.driver.usb.AppleUSBEHCIPCI	1.1
com.apple.driver.usb.AppleUSBEHCI	1.1
com.apple.driver.AppleIntelLpssGspi	3.0.60
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.iokit.IOAHCIFamily	288
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.DiskImages	444.20.3
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: MacBookAir6,2, BootROM MBA61.0099.B04, 2 processors, Intel Core i5, 2.49 GHz, 4 GB, SMC 2.13f9
Graphics: Intel HD 4600, Intel HD 4600, Built-In
Memory Module: BANK0/DIMM0, 4 GB, DDR3, 1600 MHz, Hynix Semiconductor, HMT451S6AFR8A-PBN0
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: HL-DT-ST DVDRAM GT80N
Serial ATA Device: ST500LT012-1DG142, 500.11 GB
USB Device: USB 3.0 Bus
USB Device: BCM43142A0
USB Device: USB Optical Mouse
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB 2.0 Bus
USB Device: Hub
Thunderbolt Bus: 
```

So I cannot shutdown my computer normally, hope someone could solve my problem😀.

usr-sse2, 2016-11-15:
Don't use the outdated 1.8.12 kext, current version is 1.8.24.

RehabMan, 2016-12-07:
You're using the wrong version.  Must use 1.8.24 or later with Sierra.  Closing due to user error.

closed 2016-12-07 by RehabMan