RehabMan/OS-X-Voodoo-PS2-Controller #92: Lenovo X220: scroll with middle button register additional click

kombinieren, 2016-11-13:
Configuration: Lenovo X220 with UltraNav (trackpoint + "buttonless" touchpad), El Capitan set up according to [John McDonnell installation guide](http://x220.mcdonnelltech.com/os-x-el-capitan-installation/)

_How to reproduce:_

- open a folder of files in SublimeText 3;
- select a file from tree view to display it in editor window;
- try scrolling sidebar and editor using TrackPoint:
  - press and hold UltraNav's middle button;
  - nudge the red nub up and down;

_Expected behaviour:_

- tree view in sidebar scrolls;
- editor panel scrolls;

_Actual behaviour:_

- the item in sidebar under mouse pointer recieves click and becomes "selected";
  - if it's a file, it opens for preview in editor;
- text in editor panel recieves click:
  - scroll drags cursor with it;
  - cursor selects text as it passes by it (the panel scrolls);

_Proposed solution:_
- an option to disable "click" for middle button and make it scroll-only via `.plist` file inside kext;

tluck, 2016-11-14:
@kombinieren 
for this specify issue, you may want to try the latest kext I just updated for the Lenovo T420 system - in which I may have fixed this issue of click while scrolling with TrackPoint.  Right now, the code will filter out middle clicks if MouseMiddleScroll is set to true - which means the middle button is just for scrolling support. 

you can get the compiled kext here:  http://www.insanelymac.com/forum/topic/285678-lenovo-thinkpad-t420-with-uefi-only/page-151#entry2318654

repo: https://github.com/tluck/OS-X-Voodoo-PS2-Controller (fork from this repo)

kombinieren, 2016-11-27:
Looks promising. Will test it when I'll get home, then will report results. Thank you!

RehabMan, 2016-12-07:
Won't fix thinkpad specific issues here.  Closing.

closed 2016-12-07 by RehabMan