RehabMan/OS-X-Voodoo-PS2-Controller #93: Peculiar Brightness shortcut on Sierra

MuntashirAkon, 2016-11-16:
Today, while working in a project, I accidentally pressed `Fn+B` and brightness increased. After trying out all the possible keys, I also found that `Fn+B` decreases brightness.

Jenny21David, 2016-11-16:
For now, you can use the following keys to change your brightness in Sierra 10.12.x: 
1. Pause break = Increase Brightness
2. Scroll Lock = Decrease Brightness.

I'm assuming you are using Voodoops2controller.kext version 1.8.24.

MuntashirAkon, 2016-11-16:
Yes, latest version with `ApplePS2SynapticsTouchPad` on Sierra 10.12.dp1.

tluck, 2016-11-16:
a lot of things with keyboards/PS2 (the so-called NX and special keys etc)  are broken/messed up in the early versions of sierra. i am on 10.12.2 beta 3 for that reason.

MuntashirAkon, 2016-11-17:
Same situation after updated to macOS Sierra 10.12 beta 8.

MuntashirAkon, 2016-11-17:
Another odd thing is that pressing `Fn+R` prints or does nothing in normal situation, but it prints `~` if trying it in the Terminal.

MuntashirAkon, 2016-11-19:
Updated to 10.12.1 and the issue persists.

MuntashirAkon, 2016-11-27:
Also, after updating to 10.12.1, I found that if I pressed `Fn+F12`, brightness increased (but pressing `Fn+F11` didn't decrease brightness). Again, if I pressed `Fn+S` (or, `Scroll Lock`), it didn't work. But pressing `Fn+B` (or `Pause Break`) worked. And after that, `Fn+F12` didn't work anymore.

I have updated to 10.12.2 beta 3, nothing changed.

x4080, 2016-12-02:
I found fn b and s as well, can we just map in the plist? What I don't know is getting the scan code to replace the fn b and s

RehabMan, 2016-12-07:
I will be updating the kext to send F14/F15 by default for the brightness keys.

closed 2016-12-07 by RehabMan