RehabMan/OS-X-Voodoo-PS2-Controller #94: How to build? (for beginner)

ryanknutson, 2016-11-17:
Hello, so how can I build this on my own? I'm trying to use the version that @usr-sse2 created (that fixes the trackpad settings window). How can I build this? I haven't done any software compiling before. Thanks!

usr-sse2, 2016-11-17:
Install Xcode
git clone https://github.com/usr-sse2/OS-X-Voodoo-PS2-Controller.git
cd OS-X-Voodoo-PS2-Controller
git checkout 3fdrag
open VoodooPS2Controller.xcodeproj
In Xcode window, select scheme 'All Kexts'
Press Command-B
Right-click Products/VoodooPS2Controller.kext in Xcode window and select Show in Finder

ryanknutson, 2016-11-17:
Thank you! Do I install Xcode 6.1 or will Xcode 8.1 work?

usr-sse2, 2016-11-17:
I used Xcode 8.1 to build for both Sierra and Snow Leopard, it works.

philippetev, 2016-11-17:
Only Xcode won't be enough. The older SDKs for Xcode are required too, because the deployment target platform is 10.6 and you won't be able to compile the kext without them. I've collected all those SDKs here:
https://mega.nz/#!lFtECDaA!HJCsGPm3MHerufjeXJ91DNEd1ZCkopLPcStHWVccLhw
Just extract them to /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs or extract them to any other place and make symlinks to them in the above path.

usr-sse2, 2016-11-17:
I set the deployment platform to 10.12, used the current (10.12) SDK and the kext compiled and even appeared to be binary compatible with 10.6.

ryanknutson, 2016-11-17:
Alright cool.

tluck, 2016-11-17:
FWIW... lately, I have been compiling with Xcode 8.1 using 10.12 SDK and target 10.12 (instead of 10.6). The kext works on either El Cap and Sierra - as those the OS versions that i care about.

x4080, 2016-11-30:
@usr-sse2 Thanks for the tip, was looking for this :)

x4080, 2016-12-01:
I have problem with compiling, the result kext is only 81kb, the kext downloaded rom @tluck  is 296kb, anybody has any tips?

xcode 8.1, sdk 10.2

RehabMan, 2016-12-07:
Closing.  There is no issue building these kexts if you have the correct Xcode setup.

closed 2016-12-07 by RehabMan