RehabMan/OS-X-Voodoo-PS2-Controller #96: Synaptics touchpad tracking issues (usr-sse2)

Petapton, 2016-11-22:
Hi, I'm using usr-sse2 version of the kext.
My trackpad is a Synaptics without separate buttons and while I'm dragging (using physical button) pointer goes down left, as trackpad tracks a finger in the bottom left corner. The only way to make it work is to press button with a nail, so trackpad registers click, but doesn't see the finger. Any help? Thanks

usr-sse2, 2016-12-07:
I have such trackpad too, and such dragging never worked for me with RehabMan's version. Do I need to enable some option?

RehabMan, 2016-12-08:
Dragging works fine here.  In 10.12, you enable it in SysPrefs->Accessibility.

closed 2016-12-08 by RehabMan

usr-sse2, 2016-12-08:
This discussion is not about dragging with one finger, which works, but about dragging with two fingers – one holding the button and another moving the cursor.

Strange – it sometimes works and sometimes doesn't, will check in the official 1.8.25 later. Also trying to click with physical buttons has always been very unreliable because while I click, the cursor moves.

RehabMan, 2016-12-15:
It works just fine here.