RehabMan/OS-X-Voodoo-PS2-Controller #103: Mapping Caps Lock to Command

bel3atar, 2016-12-11:
Results in command being pressed all the time.
To reproduce: System preferences => Keyboard => Modifier Keys

RehabMan, 2016-12-12:
I assume you're on 10.12.
If so, "won't fix".  This is an artifact of the capslock bug work around.

closed 2016-12-12 by RehabMan

bel3atar, 2016-12-12:
I'm 10.12.1, does that make a difference?

RehabMan, 2016-12-15:
My comment is applicable to all versions of 10.12 so far.

emdriventodrink, 2017-01-31:
Same problem here.  I understand "won't fix."  Is there another way around this?  I'm using an old IBM model M and have limited choices for a command-key.  I pretty much have to remap CAPSLOCK to something.  Thanks much!

DaiMaoTsai, 2017-01-31:
I understand "won't fix."  Is there another way around this?  I'm using an IBM model M and have limited choices for a command-key substitution.  I pretty much have to remap CAPS LOCK to *something*.  Thanks!

RehabMan, 2017-02-01:
You can remap the key with ACPI "Custom ADB Map".
See the u430 repo (and keyboard remap wiki) for ideas.

t-marvolo, 2017-07-05:
I'm trying to do the same thing as DaiMaoTsai, but can't figure out the PS2 scan codes I need. I've installed the debug kext and daemon, rebuilt caches and permissions and rebooted but I don't know where I'm supposed to see the scan codes.

Any help would be appreciated.

DaiMaoTsai, 2017-07-05:
I wound up installing karabiner elements and used it to do the remapping.

https://github.com/tekezo/Karabiner-Elements

t-marvolo, 2017-07-06:
Yeah I tried with karabiner elements too, but was having the same issue with the key sticking. 

If I watch with the karabiner event viewer, while caps is mapped to control for example, the release event happens BEFORE the press event (where it gets stuck). Pressing the actual control key then frees it from being stuck again. 

How do you have yours configured exactly?

DaiMaoTsai, 2017-07-06:
Just one single remap does it for me.

[image]

t-marvolo, 2017-07-06:
Okay and you have the mapping in system prefs default? It seems karabiner overrides that anyway, but I'm not sure why I can't get it working. I guess I should just pick up a ps/2 to usb converter at this point.

DaiMaoTsai, 2017-07-06:
Yes, I use the defaults and everything works great.  

[image]

PS/2 to USB is probably a good idea since these devices will probably be orphaned at some point, even though mine shows no sign of aging.  Also Unicomp (original MFGR of the Model M) [makes new ones](http://www.pckeyboard.com/page/category/SpacesaverM) with MacOS keys.  Good luck

t-marvolo, 2017-07-06:
Picked up the converter. It works fine now, but only with Karabiner. The system pref remapping doesn't seem to have any affect with the converter.

I've seen the bluetooth and usb mods around, maybe I will check them out one day.

plumps, 2017-08-07:
As a follow-up for somebody else struggling with Karabiner Elements and a sticky CapsLock key. Here is what I did. 

```bash
diskutil mount /dev/disk0s1
open /Volumes/EFI/EFI/CLOVER/kexts/other/VoodooPS2Controller.kext/Contents/PlugIns/VoodooPS2Keyboard.kext/Contents/Info.plist
```
And added the following key  to *IOKitPersonalities>ApplePS2Keyboard>Platform Profile>Default>Custom ADB Map*
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<string>3a=3d;swap caps_lock with right_option</string>
</plist>
```

4volodin, 2020-03-21:
> As a follow-up for somebody else struggling with Karabiner Elements and a sticky CapsLock key. Here is what I did.
> 
> ```shell
> diskutil mount /dev/disk0s1
> open /Volumes/EFI/EFI/CLOVER/kexts/other/VoodooPS2Controller.kext/Contents/PlugIns/VoodooPS2Keyboard.kext/Contents/Info.plist
> ```
> 
> And added the following key to _IOKitPersonalities>ApplePS2Keyboard>Platform Profile>Default>Custom ADB Map_
> 
> ```
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <string>3a=3d;swap caps_lock with right_option</string>
> </plist>
> ```

hi bro, did you have the problem like here ? https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/issues/194