RehabMan/OS-X-Voodoo-PS2-Controller #110: Intercept special key combination

RRPRIME, 2016-12-30:
Hello,

I am faced with the following scenario:

I have a asus b551 I am trying to get the FN+F10, F11, F12 input to the system
ApplePS2Keyboard debug does not show any code for this
ACPIDebug does to show any EC call or other calls

Any ideas how to intercept this ?

Cheers

RehabMan, 2016-12-31:
Mute/Vol+/Vol- are always handled via PS2 (they have standard PS2 codes).
That's why they can be translated to Mute/Vol+/Vol- ADB codes automatically and for every system.
Brightness keys (and other keys) are different in that they either have non-standard PS2 codes, or no PS2 codes at all (handled via ACPI).
Keep in mind the new logging mechanism in Sierra.

closed 2016-12-31 by RehabMan

RRPRIME, 2016-12-31:
Mute/Vol+/Vol- are always handled via PS2 (they have standard PS2 codes).
--- Then why aren't they generating key codes via PS2 debug ?
That's why they can be translated to Mute/Vol+/Vol- ADB codes automatically and for every system.
--- Irrelevant at this point
Brightness keys (and other keys) are different in that they either have non-standard PS2 codes, or no PS2 codes at all (handled via ACPI).
--- I clearly stated "ACPIDebug does to show any EC call or other calls"
Keep in mind the new logging mechanism in Sierra.\
--- What new logging system ? I am using your latest version build https://bitbucket.org/RehabMan/os-x-voodoo-ps2-controller/downloads/RehabMan-Voodoo-2016-1208.zip in debug mode and it is not generating any codes other keys work OK

Any ideas how to intercept this ?

RehabMan, 2016-12-31:
> --- Then why aren't they generating key codes via PS2 debug ?
Probably you are not gathering logs kernel logs correctly.  If this is Sierra, there is a new logging mechanism you must be aware of.
> --- Irrelevant at this point
Not irrelevant.
> --- I clearly stated "ACPIDebug does to show any EC call or other calls"
You are probably not looking at the logs correctly (Sierra has a new mechanism)
Read about it on developer.apple.com.
Or read my laptop FAQ on tonymacx86.com
But you should not expect to see EC methods called due to the volume control keys.  They are always PS2.