RehabMan/OS-X-Voodoo-PS2-Controller #111: BUG: NO PS2 codes for FN+F keys

RRPRIME, 2016-12-31:
OS: 10.12.2
System: Asus b551 (laptop) 
Bug: FN+F10, F11, F12 (mute, vol up, vol down) via ApplePS2Keyboard debug does not show any code 

Steps to repro: Install Voodoo-PS2-Controller Debug version, use ioio -s ApplePS2Keyboard LogScanCodes 1 to trigger the logging of the codes watch dmesg.

Keys output normal, example:
ApplePS2Keyboard: sending key f=30 down
ApplePS2Keyboard: sending key f=30 up

F10, F11, F12 output codes but FN+F10 or +F11 or +F12 does not output anything

RehabMan, 2017-01-01:
No bug in the kext.  Your mistake in trying to collect kernel logs the wrong way...

There is a new logging mechanism in Sierra.  Read at developer.apple.com

closed 2017-01-01 by RehabMan

RRPRIME, 2017-01-01:
there is no "new way", all your output is in dmesg and you can see by the example I posted that I have the correct information and the location to collect it.

RehabMan, 2017-01-01:
New mechanism is 'log'.  Evidently dmesg, although deprecated, is still working.. (dmesg a bit inconvenient to use though).

Your keys may be handled some other way (or you're pressing the wrong keys).
Or you may have patched ACPI incorrectly (the EC may respond differently regarding the keyboard depending on Windows version it thinks is running).
At any rate, no bug in this kext... it is some troubleshooting you must do for your specific computer.
If a key generates a PS2 code (by interrupt/PS2 registers), you will see it in the debug output.  There is nothing in the code that specifically filters out particular PS2 keyboard input as coming from the i8042 keyboard controller.

The issues database in github is not the appropriate place to troubleshoot your system specific problem.