RehabMan/OS-X-Voodoo-PS2-Controller #112: BUG: Trackpad going crazy after install

RRPRIME, 2016-12-31:
OS: 10.12.2
System: Asus b551 (laptop)
PS2M object from DSDT

Scope (_SB.PCI0.LPCB)
    {
        Device (PS2M)
        {
            Name (_HID, EisaId ("ETD0108"))
            Name (_CID, Package (0x05)
            {
                EisaId ("SYN0300"), 
                EisaId ("SYN0302"), 
                EisaId ("SYN0002"), 
                EisaId ("PNP0F13"), 
                EisaId ("PNP0F03")
            })
            Name (_CRS, ResourceTemplate ()
            {
                IRQ (Edge, ActiveHigh, Exclusive, )
                    {12}
            })
            Method (_STA, 0, NotSerialized)
            {
                If (LEqual (TPME, Zero))
                {
                    Return (Zero)
                }

                Return (0x0F)
            }
        }
    }

ioreg:
http://www.mediafire.com/file/xz8dlryed3qkrtp/sajuuk%E2%80%99s_MacBook_Pro.ioreg

BUG: After the installation of the kext in EFI clover partition on boot the trackpad is going crazy (random jumps, random clicks even gets the USB mouse stuck on some buttons in pressed state)

Steps to repro: Install latest kext, boot.

Workaround:  remove VoodooPS2Mouse.kext from the main VoodooPS2Controller.kext reboot then add the VoodooPS2Mouse.kext back and reboot, it should behave normally now

RehabMan, 2017-01-01:
This kext has no ELAN support.

closed 2017-01-01 by RehabMan

RRPRIME, 2017-01-01:
Well, excuse me for tying to get a bug fixed, I know you have no support for Elan but since there is a workaround it means you have a bug and you can spend 10-15 min identifying the issue.

RehabMan, 2017-01-01:
The ELAN trackpad firmware has poor support for mouse emulation.
Find a PS2 kext for ELAN (you will find it is not open source).
If I ever get a laptop with ELAN, I will write specific support in my kext.  But since I don't have any such hardware, I'm not able to write any software for it.