RehabMan/OS-X-Voodoo-PS2-Controller #113: BUG: NO PS2 codes for FN+F keys

RRPRIME, 2017-01-01:
OS: 10.12.2
System: Asus b551 (laptop)
Bug: FN+F10, F11, F12 (mute, vol up, vol down) via ApplePS2Keyboard debug does not show any code

Steps to repro: Install Voodoo-PS2-Controller Debug version, use ioio -s ApplePS2Keyboard LogScanCodes 1 to trigger the logging of the codes watch dmesg.

Keys output normal, example:
ApplePS2Keyboard: sending key f=30 down
ApplePS2Keyboard: sending key f=30 up

F10, F11, F12 output codes but FN+F10 or +F11 or +F12 does not output anything

After more investigation FN+any F key beside F5, F6, F8 (brightness up (calling EC), bright down (calling EC), mirror(calling EC)) does not generate any PS2 code

**All keys are outputted in dmesg beside the combination I presented. There is no new way of logging in Sierra there is only one way of logging as it always was.**

**You either have:
1) BUG: you are not logging all the information in the same place which is a bug
2) BUG: your kext is not logging all the information and you missed a case somewhere which is a bug
3) hardware issue - highly doubt it**

RehabMan, 2017-01-01:
No bug here.  See duplicate report #111

closed 2017-01-01 by RehabMan

RehabMan, 2017-01-01:
Bottom line: If there is valid PS2 input generated by the i8042 PS2 keyboard controller/EC, it will be in the debug output.  No output indicates no valid PS2 input.