RehabMan/OS-X-Voodoo-PS2-Controller #114: TrackPoint Only Recognized 20% of the Time

cfis, 2017-01-16:
I've been using the VoodooPSController for quite a few years now, currently on a Thinkpad t450s.

The extension has worked fine on previous OSX releases such as Mountain Lion, El Capitan, etc. 
But starting with Sierra, the track point only works about 1 out of 5 times I boot the machine.  The trackpad always works correctly.  By not working, I mean using the trackpoint does nothing (mouse cursor doesn't move).  Its like the hardware isn't being recognized at boot up.

To solve I just reboot over and over until it does work.  I haven't been able to detect a pattern (say boot to windows then osx, boot to just osx, restart/versus shutdown/start, having a iphone plugged into a usb port, etc).

Anyone see something similar and any hints on how to approach fixing it?

RehabMan, 2017-01-17:
You should probably do some debugging.  I'm not able to as I have no hardware with trackpoint.

cfis, 2017-01-17:
Yeah, nothing shows up in the system log with a debugging build.

tluck, 2017-01-17:
i have good news other that misery loves company. I have/had the same issue on the T420 for quite a long time (yosemite?). cant figure out the issue. i.e Randomly the TrackPoint and buttons are dead on cold start. a sleep/wake will always bring it to life. The same kext works without any problems on my T460 - TP works every start.

tried:   
removed the other Trackpad types
remove ps2mouse.kext
SSD vs HD - timing 

it seems totally totally random. i get a failure about 1 of 5 times. but i put it to sleep and wake it vs restart which is iffy.

cfis, 2017-01-17:
Interesting...I should probably get sleep/wake working and try that route.

Note that once it happens, unloading and reloading the kext makes no difference.  Its seems as if the OS is simply not recognizing the trackpoint exists at all.  I'll see if there is something different in the output from IO Registry Explorer between working and not working.

FYI @tluck I saw you have updated the driver for your Thinkpad - cool.  I gave it a try a month ago but have had better luck with an updated version of the changes that jcsnider did (although those changes are fairly hacky).  See https://github.com/cfis/OS-X-Voodoo-PS2-Controller/commits/lenovo branch.  Will see if I can get your version working better for me one of these days.

RehabMan, 2017-01-17:
Keep in mind there is nothing in the OS or the kext that needs to "recognize the trackpoint".

The trackpoint data is mixed with the trackpad data by the EC/trackpoint/trackpad firmware.

tluck, 2017-01-17:
yeah. kext unload/load has no effect for me either. a sleep/wake does provide function very consistently. 
my branch has some changes to the Synaptics driver that are specific to T460 with its new ClickPad and 3 physical buttons for TrackPoint. the big difference seems to be the actual buttons are not pass through and needed to be decoded differently.  the code currently is based on 1.18.25. vs the jcsnider was pretty old last i checked so had issues on Sierra. oh yeah  - trackpoint scrolling and cut/paste works too. anyway the T460 kext may work on T450?

kenthinson, 2017-01-23:
I'm having the same issue with Thinkpad x230. If I turn off the trackpad in bios then the touchpoint works 100%. But I would really like to be able to use both.

tluck, 2017-01-24:
i dont know if the 230 is the same as the 420. but i changed the initialization in the TrackPad kext to do a full hw reset and now the ThinkPad T420 (synaptics v7.2) has been consistent and reliable to enable the TrackPoint on startup. so now no more reboots or sleep/wake to get both trackpad and track point. wonderful

cfis, 2017-01-25:
@tluck - that is interesting, that does seem to make it better.  Not perfect, but its at least flipped from 20% \ 80% to 80% \ 20%.

kenthinson, 2017-02-07:
@tluck what did you do to get it working? thanks!

kenthinson, 2017-02-16:
@cfis can you give me details on what to change to make it do the hw reset? Thanks!

tluck, 2017-02-17:
simple. I changed the code to enable the full hw reset upon initialization in addition to enabling "undocumented" features.

kenthinson, 2017-02-20:
@tluck Can you be more specific with the way you did this? what files did you edit? Were they just the plist files or did you create a whole new recompile. thanks! :)

tluck, 2017-02-20:
I forked a branch and and among the various c++ code changes - this particular change was in the source file VoodooPS2SynapticsTouchPad.cpp -  in the section for init options.

kenthinson, 2017-03-16:
@tluck Thanks man! I compiled using your repo and now my x230 works 100%

RehabMan, 2017-05-13:
No plans to do anything regarding trackpoint here, unless someone submits a concise/well written pull request.

closed 2017-05-13 by RehabMan

mikethebos, 2018-10-20:
@RehabMan @tluck @cfis @kenthinson This has been fixed by @jcsnider. See kext in post 25 here:  https://www.tonymacx86.com/threads/guide-lenovo-thinkpad-t440s.244798/page-3#25

This needs to be changed:
Prior to this block of code running you are setting skipassthru=1 and passthru=1.

```
    if (!skippassthru)
    {
        UInt8 passthru2 = buf3[2] >> 7;
        // see if guest device for pass through is present
        UInt8 passthru1 = 0;
        if (getTouchPadData(0x1, buf3))
        {
            // first byte, bit 0 indicates guest present
            passthru1 = buf3[0] & 0x01;
        }
        // trackpad must have both guest present and pass through capability
        passthru = passthru1 & passthru2;
#ifdef SIMULATE_PASSTHRU
        passthru = true;
#endif
        DEBUG_LOG("VoodooPS2Trackpad: passthru1=%d, passthru2=%d, passthru=%d\n", passthru1, passthru2, passthru);
    }
```
The problem is that there is another function overriding the skippassthru variable, so that block of code runs anyways. See function setParamPropertiesGated in the VoodooPS2SynapticsTouchPad.cpp. This line specifically is what is overriding your skippassthru value from one of the plists I believe:

`        {"SkipPassThrough",                 &skippassthru},`
Anyways, my personal "quick fix" was to do the following.
Add:
```
skippassthru = 1;
passthru = 1;
```
Above:
`if` (!skippassthru)