RehabMan/OS-X-Voodoo-PS2-Controller #115: key is not mapping correctly in Thinkpad X1 yoga(OSX 10.12.3 )

xinyuwufei, 2017-02-05:
Hi,Rehabman

I both tried RehabMan-Voodoo-2016-1208.zip and RehabMan-Voodoo-2016-0616.zip on OSX 10.12.3
keys like
 '[' ,
 ']' , 
'\' and also some keys with the 'Shift' key pressed, are not mapping correctly on my thinkpad x1 yoga laptop.

it works out of box in osx 10.11.X.
any idea to solve this?
thanks in advance!

RehabMan, 2017-02-05:
What do you mean by "not mapping correctly"?
Use the debug version to get logs.

xinyuwufei, 2017-02-05:
hi Rehabman,

now i'm using the debug version, I mean the keyboard is typing wrong character(wrong position)
eg:.

key | mapping output | debug
--- | --- | ---
[ | @ | key 1a=21
] | [ | key 1b=1e
/ | ] | key 2b=2a

RehabMan, 2017-02-05:
the 1a=21, 1b=1e, 2b=2a are as expected and work correctly with my keyboard here.
You probably have a non-US keyboard/setup.
Use Ukelele to fix or visit SysPrefs->Keyboard->Input Sources.

closed 2017-02-05 by RehabMan