RehabMan/OS-X-Voodoo-PS2-Controller #117: JP106 keyboard ID

since19861019, 2017-02-26:
VoodooPS2Controller IOMatch has the following key  for KBC.
<string>PNP0303</string>

Some Japanese laptops having keyboard 106 returns "PNP0320" as HID and I have to patch DSDT.
Will you add the next value in the Info.plist ? 
<string>PNP0320</string>

I have added the value and confirmed that it works.

RehabMan, 2017-03-22:
Attach DSDT.aml for verification.

since19861019, 2017-03-23:
[DSDT.aml.zip](https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/files/863320/DSDT.aml.zip)

RehabMan, 2017-03-23:
Fixed.  Will be in a future build.

closed 2017-03-23 by RehabMan

do-n-khanh, 2017-05-26:
Could you please release a build as soon as possible.
"future build v1.8.26
    Now matches on PNP0320 (Japanese keyboard). Fix for issue #117"

I dont have experience, I just know to download the build and put it on Kext Utility