RehabMan/OS-X-Voodoo-PS2-Controller #119: touchPoint move too fast

zhiwei-lin, 2017-04-06:
**Appreciate your great work!**

on my x230t  touch pad and nav key runs very well,
but the touch point moves too fast, tough to anchor to a position, and changing mouse speed in syspref doesn't work.
a minor issue is, I want to disable touch pad when using touch point, low priority through.

tluck, 2017-04-06:
you may want to look at the Lenovo T420/T460 fork by tluck (me) which has some TrackPad and TrackPoint settings that work well this generation Synaptics. PrtSc halts/toggles the TrackPad and one continue to use TrackPoint.

g1204, 2017-04-12:
thx. I tried tluck's Voodoo-PS2-Controller, but track point (along with 3 buttons in track pad) doesn't work on my x230t with sierra 12.4.

PrtSc does work. 

so for now i will only use track pad with the old original Voodoo-PS2-Controller.
two-finger scrolling gesture works well.

btw, is it possible to enable the touch screen functionality on my x230t, considering macbook doesn't support this.

RehabMan, 2017-05-13:
No plans to do anything about trackpoint here, unless someone submits a well contemplated pull request.

closed 2017-05-13 by RehabMan