RehabMan/OS-X-Voodoo-PS2-Controller #120: SynapticsTouchPad cant zoom（Just like magnify the picture）

dgtutu, 2017-04-10:
Why the VoodooPS2Controller.kext  can't  make my SynapticsTouchPad zoom in Hackintosh(10.12.x)（Just like magnify the picture），Can you fix it ，Thx ：）

tluck, 2017-04-10:
Zoom (full screen or PIP) works for my Synaptics Touchpad ? did you enable in Accessibility?

dgtutu, 2017-04-10:
eh ？Where can ienable it ，info.plist  or touchpad setting？I am sure I had enabled it in touchpad setting

tluck, 2017-04-10:
are you using latest VoodooPS2 for Sierra support?  nothing to change in kext/info.plist that i know of. 
1) you enable scrolling in TrackPad SysPref. 
2) then in sys-pref Accessibility - enable Control + Scroll to Zoom full screen.

dgtutu, 2017-04-10:
Yes，I‘m using latest VoodooPS2 for Sierra support
All in the trackpad  syspref what I can see that I had enable ，but I still can’t zoom

dgtutu, 2017-04-10:
My trackpad syspref  at the very start （I just reinstall the system） which is not shown anything(it's white) in Preference setting，Only I had installed Acpibatterymanager can I change the setting in it

dgtutu, 2017-04-10:
I download VoodooPS2 from github and  I Compiled it with xcode by myself，and put it into clover/kext/10.12 ，now I can use my two finger to roll ，but only zoom I can't use it

tluck, 2017-04-10:
test zoom with USB mouse - which has nothing to do with VoodooPS2Controller.

dgtutu, 2017-04-10:
‘test zoom with USB mouse - which has nothing to do with VoodooPS2Controller.’
I use Roller + command/ctrl /^  in my  mouse can‘t zoom anything

tluck, 2017-04-10:
did you enable zoom in Accessibility?

dgtutu, 2017-04-10:
Yes。。。

tluck, 2017-04-10:
well if you cannot zoom with USB mouse - then it is not a problem with VoodooPSController.

i have an Apple Magic mouse, USB mouse, and Synaptics TrackPad (via VoodooPS2Controller) - all work the same with Control+Scroll.

dgtutu, 2017-04-10:
Follow your advice ，I can zoom with Control+Scroll in the Pictures app，but the xxx.jpg in finder still can’t zoom it，it  will only move

dgtutu, 2017-04-10:
I opened the amplification through the auxiliary function, but it is only the entire screen to enlarge, I just want to enlarge a picture like the windows system。。。。。。。。

dgtutu, 2017-04-10:
I want my trackpad like command + ‘+’

tluck, 2017-04-10:
you can use PIP (small zoom window) - instead of full screen zoom
or use file preview feature - select an image file in finder - hit space bar. 
but this not the forum to learn macOS

dgtutu, 2017-04-10:
[image]
Why can't I use two fingers to zoom in, but I have to add a command key, like this

dgtutu, 2017-04-10:
[image]
I don't want this way+ command to zoom......
OH ,which make me sadlly

dgtutu, 2017-04-10:
[image]

I've got all the options

tluck, 2017-04-10:
ah - well 2 finger Pinch/Expand to Zoom is not implemented in the software driver. so is not a working gesture.

dgtutu, 2017-04-10:
So I just want Dear Mr. rehabman add the function....in the software driver。。。。

RehabMan, 2017-05-13:
No plans to implement pinch/zoom.

closed 2017-05-13 by RehabMan