RehabMan/OS-X-Voodoo-PS2-Controller #121: TrackPoint not working on Thinkpad x230

dallinac, 2017-05-01:
I've tried installing the kext and daemon multiple times, but I can't get the TrackPoint or the buttons above my trackpad to work. However, the TrackPoint does work in Clover when choosing which OS. I'm on El Capitan

RehabMan, 2017-05-13:
Trackpoint is not well supported with this kext (I don't have one).
Try tluck's fork.

closed 2017-05-13 by RehabMan

dustojnikhummer, 2017-08-17:
I may be stupid,but where can I download that X230-modified kext/driver? I can only see downloads for your original voodooPS2 that does not work for me (for Trackpoint and trackpoint buttons)

RehabMan, 2017-08-17:
You should look at tluck's fork.
I'm not providing downloads for other people's forks.

MachOne1976, 2018-07-29:
I fixed this problem in a very simple way in reality, I have the trackpad and the trackball that works permanently thanks to a little tip. For this, when you start the computer, when you are on the macOS selection screen just use the trackball to select the boot drive and validate with the button. Attention, do not use the trackpad at all, you do not touch it, you only use the trackball to choose the macOS drive, then once in the system you can use the one you like.﻿

mikethebos, 2018-10-20:
@RehabMan @tluck @MachOne1976 @dallinac @dustojnikhummer @cfis @kenthinson This has been fixed by @jcsnider. See kext in post 25 here:  https://www.tonymacx86.com/threads/guide-lenovo-thinkpad-t440s.244798/page-3#25

See [pull request](https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/pull/170#issue-224441115)

This needs to be changed:
Prior to this block of code running you are setting skipassthru=1 and passthru=1.

```
    if (!skippassthru)
    {
        UInt8 passthru2 = buf3[2] >> 7;
        // see if guest device for pass through is present
        UInt8 passthru1 = 0;
        if (getTouchPadData(0x1, buf3))
        {
            // first byte, bit 0 indicates guest present
            passthru1 = buf3[0] & 0x01;
        }
        // trackpad must have both guest present and pass through capability
        passthru = passthru1 & passthru2;
#ifdef SIMULATE_PASSTHRU
        passthru = true;
#endif
        DEBUG_LOG("VoodooPS2Trackpad: passthru1=%d, passthru2=%d, passthru=%d\n", passthru1, passthru2, passthru);
    }
```
The problem is that there is another function overriding the skippassthru variable, so that block of code runs anyways. See function setParamPropertiesGated in the VoodooPS2SynapticsTouchPad.cpp. This line specifically is what is overriding your skippassthru value from one of the plists I believe:

`        {"SkipPassThrough",                 &skippassthru},`
Anyways, my personal "quick fix" was to do the following.
Add:
```
skippassthru = 1;
passthru = 1;
```
Above:
`if` (!skippassthru)`

tluck, 2018-10-20:
so it seems that if one sets both SkipPassThrough and Forcepassthru to true (either by Info.plist or by SSDT property) - thats should resolve to what we need - passthru=true. what am i missing?

mikethebos, 2018-10-20:
@jcsnider any insight for @tluck?

jcsnider, 2018-10-20:
@mikethebos my changes were all merged into this repo with this pr here (https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/pull/151). I am now using the default kext from Rehabman with my Thinkpad.

As far as I know, as long as you are using the correct SSDTs it should work:
https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/blob/master/SSDT-Thinkpad_Clickpad.dsl

OR

https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/blob/master/SSDT-Thinkpad_Trackpad.dsl

tluck, 2018-10-20:
exactly.  i have similarly abandoned my branch as all the bits were merged back into RehabMan's main branch. 

i used the SSDT method to set/override any values. the samples are patterned for T420 which should be very close if not the same for any of the ThinkPads before the 440.   

https://github.com/tluck/Lenovo-T420-Clover/blob/master/EFI/CLOVER/ACPI/SSDT-T420-PS2K.dsl