RehabMan/OS-X-Voodoo-PS2-Controller #122: Synaptics Trackpad recognized as mouse

SebastianBar, 2017-05-03:
Hi!

I'm currently using the latest release of this kext, and on Sierra, my trackpad is not being recognized as a trackpad, but instead as a mouse.

On Yosemite, the trackpad is being detected with no issues at all.
Is there something I can do to fix this behavior? For example, how can I use the "ForceSynapticsDetect" feature?

Thanks!

RehabMan, 2017-05-13:
Use the forums, attach "Problem Reporting" files as per FAQ.
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/

closed 2017-05-13 by RehabMan

bySabi, 2017-12-10:
@SebastianBar do you solved it?
sames here

SebastianBar, 2017-12-10:
Nah, despite the fact that I could replicate that issue many times from clean installations of Sierra and show proof of the issue, the answer I got from RehabMan was "do a clean installation, the kext has no issues" and that was all.

I just stopped using the kext and started using a wired mouse.

bySabi, 2017-12-10:
I'm on a High Sierra laptop, synaptics touchpad works perfectly fine with ElCapitan but not now.

I see that VoodooPS2Daemon not load from LaunchDaemos but I can start it from console with`/usr/bin/VoodooPS2Daemon`

What is your hardware?
Do you see VoodooPS2Daemon on Activity Monitor??

bySabi, 2017-12-10:
Well, kext is working fine! 
`tap` works before user login but stop on login ... It seem VoodooPS2Daemon launchd stop at this point.
@RehabMan  any idea?

bySabi, 2017-12-10:
kext is working fine and VoodooPS2Daemon is up but still not TrackPad pref pane.
I set tap and some gestures with this script: https://github.com/boochtek/mac_config/blob/master/trackpad.sh

RehabMan, 2017-12-11:
Use the forums:
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/