RehabMan/OS-X-Voodoo-PS2-Controller #127: VoodooPS2Daemon does not work on HighSierra macOS or another files.

jonathanburgossaldivia, 2017-07-04:
It works correctly in macOS Sierra, you see the preferences panel of the Trackpad as the NATIVE, unfortunately no longer works in this version of macOS HighSierra, probably need an update one of your files, i have prepared some shell scripts in plain text for you to install all files (kexts, demons, and others), maybe you can help me to correct something, I regret to have written you here since I do not have your mail, for thank you very much for all the support you offer for PCs, this is the link where my files are hosted for you: https://drive.google.com/drive/folders/0B3nXjlaODkBvS21wSTFIOWVuWFU?usp=sharing   .

RehabMan, 2017-08-10:
Duplicate of #125

closed 2017-08-10 by RehabMan

ykla, 2017-08-17:
I have the same issue.

bySabi, 2017-12-10:
@ykla @jonathanburgossaldivia do you solved it??

sames here

RehabMan, 2017-12-11:
Use the forums.
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/

jonathanburgossaldivia, 2017-12-11:
Is working again in macOS highsierra, works perfect now ;)

bySabi, 2017-12-12:
@jonathanburgossaldivia do you have Trackpad pref panel?

RehabMan, 2017-12-12:
This bug is already fixed.
And it has nothing to do with a Trackpad prefpane.
Use the forums.