RehabMan/OS-X-Voodoo-PS2-Controller #128: How to disable unwanted features in Trackpad

favadi, 2017-07-17:
My machine is a Dell Precision 5510 and while it works, it doesn't work great. The 2 finger scrolling feature feel laggy, somehow 2-3 times out of 10, it doesn't move. I suspect that it might incorrectly recognizes as 2 finger taps or something else.

So the question is: How can I disable all actions/gestures except left click/right click/2 finger scrolling?

There are too many options in Info.plist and I'm not sure what they meant.

scottsanett, 2017-08-06:
Try ApplePS2SmartTouchpad.kext from my repo. [link](https://github.com/scottsanett/M5510-4K-Sierra-Installation/tree/master/CLOVER/kexts/10.12/ApplePS2SmartTouchPad.M5510.kext/Contents)

favadi, 2017-08-07:
Actually, I found out that I was using https://github.com/syscl/OS-X-Voodoo-PS2-Controller, after switching to this version, I don't have those gestures features anymore and the 2 finger scrolling feature works more reliable. I will try your kext if having any problem in the future. Closing this.

closed 2017-08-07 by favadi