RehabMan/OS-X-Voodoo-PS2-Controller #131: Driver not loading on PNP0303/DLLK062B since 5b62873

OttoEdler, 2017-08-04:
The trackpad on my Dell Latitude E5450 doesn't work with the latest binary (July 22). In IORegistryExplorer there is no driver loaded on PS2M device. It used to work with earlier versions.

RehabMan, 2017-08-10:
Use the forums to provide "Problem Reporting" files.
Read FAQ, "Problem Reporting"
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/

OttoEdler, 2017-08-10:
I don't use tonymacx86.com. Should I attach ioreg with working trackpad drivers or with your latest binary?

RehabMan, 2017-08-10:
I'm not providing support here.
Only on tmx.
Your choice.

closed 2017-08-10 by RehabMan

OttoEdler, 2017-08-10:
Why do you spread your projects across 3 platforms? I don't see a good reason, since all essential features are available here.

RehabMan, 2017-08-10:
No idea what you mean by "3 platforms".

OttoEdler, 2017-08-10:
github.com, bitbucket.org and tonymacx86.com

RehabMan, 2017-08-10:
github is used as primary source code repo.
But github provides no download facility for builds (it used to, then it was discontinued).
Hence downloads are on bitbucket.
bitbucket also serves as a redundant location for source code.
tonymacx86.com is for support.

OttoEdler, 2017-08-10:
release page is the place to provide builds for download, this issue page is for reporting, organization and support. i still don't see a reason to use those forums, that are in focus of criticism for good reason. but fine, your choice, i'm on in a position to argue with that. at least you know that there is an issue with the latest version of this project.

RehabMan, 2017-08-10:
Release page did not exist when I started on github.
I'm not going to change now.  No benefit.

There is no issue.  I'm using the latest build without a problem.
Please use the forums to troubleshoot your issue.