RehabMan/OS-X-Voodoo-PS2-Controller #136: Trackpad Pane is blank in system config on High Sierra

goagain, 2017-09-29:
My laptop is HP Spectre X360, kabylake version, it use Synaptics SMBus TouchPad. The details I can got is HID\VEN_ELAN&DEV_0732&Col01.

under High Sierra, the trackpad can move , press click, double finger scroll.  but can't tap, three finger, four finger, or do other gestures.

goagain, 2017-09-29:
[image]
[image]
[image]
[image]
[image]

RehabMan, 2017-10-01:
Use the forums:
https://www.tonymacx86.com/threads/faq-read-first-laptop-frequent-questions.164990/

closed 2017-10-01 by RehabMan

bySabi, 2017-12-09:
@goagain, same here ...
you solved it?