RehabMan/OS-X-Voodoo-PS2-Controller #140: Switched keys.

drasbeck, 2017-11-12:
In the build of 2017-0901 I'm observing two keys being completely switched, i.e. pressing one key yields what the other key under normal circumstances would output and vice versa. I've marked the affected keys on the picture below. Is this behavior on purpose? And can it be changed? It's a minor nuisance when programming where the $§ and especially <> symbols are utilized. BTW: I'm using Danish layout on a ThinkPad T430.

The keys that have been switched:
[image]

If it's of any help to you, here's the keyboard viewer from the T430:
[image]

And here's the keyboard viewer using the same layout on my MacBook:
[image]

One thing I see from this is that it seems on the full size keyboard the key to the left of 'Z' (normally providing '<' and '>') seems to be missing entirely. This is also the case when I look up the layout on my Mac mini, yet the key _does_ exist as is shown in this picture:
[image]

RehabMan, 2017-11-19:
Your issue is not related to this kext.
You can change your keymap using Ukelele.

closed 2017-11-19 by RehabMan

drasbeck, 2017-12-28:
I agree on your assessment.

For me, this issue was somehow automatically resolved a couple of days after I posted. And the keyboard layout has been working as expected since then. Even through a macOS upgrade.