RehabMan/OS-X-Voodoo-PS2-Controller #142: High Sierra feature – Caps Lock to switch languages

usr-sse2, 2017-12-22:
macOS High Sierra adds a new feature – switching keyboards with Caps Lock. (It's only for system-provided layouts, no Russian BG46).
[image]
But with VoodooPS2 it's very buggy – most times double-press switches language and single-press activates Caps Lock if no keys pressed within 1 second after, but it's not always so. It should be related to the Sierra Caps Lock workaround and maybe to kCapsLockDoesLock.

RehabMan, 2017-12-24:
The bugs are caused by the capslock bug workaround.
As long as we need the workaround (hardcoded in VoodooPS2Keyboard), caps lock cannot be remapped.

RehabMan, 2018-04-23:
Capslock can be remapped with ACPI.
Refer to example: SSDT-Swap-LeftControlCapsLock.dsl

closed 2018-04-23 by RehabMan

frol, 2019-09-10:
According to Karabiner EventViewer, pressing CapsLock produces events in a wrong order: key_up -> key_down instead of key_down -> key_up.

Does the remapping CapsLock <-> LeftControl intend to help with the issue and fix the CapsLock button or is it just breaking another button instead of CapsLock?

P.S. I also use CapsLock as a layout switcher and I would like to fix this...

frol, 2019-09-10:
I have patched the issue for myself: https://github.com/acidanthera/bugtracker/issues/435#issuecomment-529858985