RehabMan/OS-X-Voodoo-PS2-Controller #152: The problem is about that the Synaptics touch board can't find the hardware.

hailluo, 2018-04-19:
Hello,RehabMan! I met a problem about to with your Voodoo-PS2-Controller,the Synaptics touchpad of my computer can't find hardware on Hackintosh,  it can be used，sensitivity of the Synaptics touchpad is very good. I tried many ways to solve it and all I failed. So I want to find you for help. Up to now, My Hackintosh system is macOS High Sierra 13.4

My Computer configuration:
              Model：Aspire E5 572G-57DW
              C P U:  Intel Core i5 4210M
        Graphics：Intel HD Graphics 4600
    Sound card:  ALC283
Network card：RealtekRTL8111
      Touchpad:  Synaptics touchpad
Thank you！

hailluo, 2018-04-21:
The Synaptics touchpad of my computer is on the USB device.

RehabMan, 2018-04-23:
This kext is a PS2 driver.  Not applicable to USB devices.

closed 2018-04-23 by RehabMan