RehabMan/OS-X-Voodoo-PS2-Controller #154: Caps Lock button backlight feedback is often wrong.

yaozaiyu, 2018-04-25:
When I was using Voodoo-PS2-Controller, I was crazy to press the Caps Lock button backlight feedback frequently wrong. But when I used ApplesmartTouchpad, all of this was normal. I hope to make a quick patch.

RehabMan, 2018-04-26:
I'm not able to reproduce the issue you're reporting. 

Please use the forums for proper troubleshooting.
There is no issue with capslock on any of the three laptops I have here.

closed 2018-04-26 by RehabMan

yaozaiyu, 2018-04-30:
Hello, is your system format APFs?

RehabMan, 2018-04-30:
I have some running APFS, others HFS+J.
But what does that have to do with capslock/ps2?

yaozaiyu, 2018-05-01:
I solved the problem，thanks！

Raymonf, 2018-05-01:
@yaozaiyu How did you solve it?

yaozaiyu, 2018-05-01:
[image]
close the tab。

Raymonf, 2018-05-01:
Ah, it must have been an IME bug for you. That option doesn't exist for me. Whatever.

[image]

yaozaiyu, 2018-05-01:
直接将中文好了，你要先添加一个输入法，在左下角的地方，然后就会出现我上面的选项了

Raymonf, 2018-05-01:
@yaozaiyu 原來如此！謝謝！

(In English: The solution is to add an official Apple input method editor to uncheck "use 'US' as the keyboard layout when caps lock is enabled". It's caused by having a Chinese IME.)