RehabMan/OS-X-Voodoo-PS2-Controller #166: Kernel panic with latest release

hieplpvip, 2018-09-25:
I get kernel panic with latest release (2018-0922), not with the previous one (2018-0506).

Strange thing is I don't get kernel panic if I boot in verbose mode.

Below is panic log:
```
Anonymous UUID:       B27D9264-BF59-D4DB-6567-DC286FC968B5

Tue Sep 25 16:45:52 2018

*** Panic Report ***
panic(cpu 0 caller 0xffffff801e58776f): Kernel trap at 0xffffff7fa1853937, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000002bd9b000, CR4: 0x00000000003627e0
RAX: 0x0000000000500000, RBX: 0xffffff80408ebf00, RCX: 0xffffff804161c580, RDX: 0xffffff804227cf10
RSP: 0xffffff81977dbeb0, RBP: 0xffffff81977dbeb0, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000008, R10: 0x0000000000000010, R11: 0x0000000000000000
R12: 0xffffff804161c5d0, R13: 0xffffff804227cf10, R14: 0xffffff804227cf10, R15: 0xffffff804161c580
RFL: 0x0000000000010246, RIP: 0xffffff7fa1853937, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff81977db980 : 0xffffff801e46c1c6 
0xffffff81977db9d0 : 0xffffff801e595274 
0xffffff81977dba10 : 0xffffff801e587544 
0xffffff81977dba80 : 0xffffff801e41e1e0 
0xffffff81977dbaa0 : 0xffffff801e46bc3c 
0xffffff81977dbbd0 : 0xffffff801e46b9fc 
0xffffff81977dbc30 : 0xffffff801e58776f 
0xffffff81977dbda0 : 0xffffff801e41e1e0 
0xffffff81977dbdc0 : 0xffffff7fa1853937 
0xffffff81977dbeb0 : 0xffffff801ea6f6d5 
0xffffff81977dbf00 : 0xffffff801ea723ab 
0xffffff81977dbf50 : 0xffffff801ea740c6 
0xffffff81977dbfa0 : 0xffffff801e41d557 
      Kernel Extensions in backtrace:
         org.rehabman.voodoo.driver.PS2Controller(1.9)[2207C797-FB9F-30CB-B3A0-D07FC4883BE5]@0xffffff7fa1851000->0xffffff7fa185bfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[95DA39BB-7C39-3742-A2E5-86C555E21D67]@0xffffff7f9ec57000

BSD process name corresponding to current thread: kernel_task
Boot args: dart=0 nv_disable=1 kext-dev-mode=1 -disablegfxfirmware 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64
Kernel UUID: 1AE5ACFD-3B6F-3D74-AD52-31F1430DBC6F
Kernel slide:     0x000000001e200000
Kernel text base: 0xffffff801e400000
__HIB  text base: 0xffffff801e300000
System model name: MacBookPro14,1 (Mac-B4831CEBD52A0C4C)

System uptime in nanoseconds: 1268327548
last loaded kext at 1243445024: com.apple.driver.AppleTopCaseHIDEventDriver	133 (addr 0xffffff7fa0fef000, size 61440)
loaded kexts:
org.rehabman.driver.FakePCIID	1.3.13
com.alexandred.VoodooI2CHID	1
com.alexandred.VoodooI2C	2.0.1
org.coolstar.VoodooGPIO	1.1
com.rehabman.driver.ACPIPoller	0.8.1
org.rehabman.voodoo.driver.PS2Controller	1.9.0
org.rehabman.driver.AppleSmartBatteryManager	1.90
com.hieplpvip.AsusSMC	1
com.hieplpvip.AppleBacklightFixup	1.0.0
as.vit9696.SMCProcessor	1.0.0
as.vit9696.VirtualSMC	1.0.0
as.vit9696.AppleALC	1.3.2
as.lvs1974.BT4LEContiunityFixup	1.1.1
as.lvs1974.AirportBrcmFixup	1.1.5
com.rehabman.driver.USBInjectAll	0.6.7
com.alexandred.VoodooI2CServices	1
as.vit9696.WhateverGreen	1.2.3
as.vit9696.Lilu	1.2.7
com.apple.driver.AppleTopCaseHIDEventDriver	133
com.apple.driver.AppleFileSystemDriver	3.0.1
com.apple.filesystems.apfs	748.51.0
com.apple.driver.AirPort.Brcm4360	1240.19.1a3
com.apple.filesystems.hfs.kext	407.50.6
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.BootCache	40
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0
com.apple.AppleSystemPolicy	1.0
com.apple.driver.AppleAHCIPort	329.50.2
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleACPIButtons	6.1
com.apple.driver.AppleBacklight	170.12.6
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleACPIEC	6.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	183
com.apple.security.TMSafetyNet	8
com.apple.security.quarantine	3
com.apple.security.LiluFriend	1.1.0
com.apple.driver.AppleHSBluetoothDriver	133
com.apple.driver.IOBluetoothHIDDriver	6.0.7f10
com.apple.iokit.IOBluetoothFamily	6.0.7f10
com.apple.driver.AppleMultitouchDriver	1404.4
com.apple.driver.AppleInputDeviceSupport	1404.3
com.apple.driver.AppleXsanScheme	3
com.apple.iokit.IO80211Family	1200.12.2
com.apple.driver.mDNSOffloadUserClient	1.0.1b8
com.apple.driver.corecapture	1.0.4
com.apple.filesystems.hfs.encodings.kext	1
com.apple.iokit.IOAHCIBlockStorage	301.40.2
com.apple.iokit.IOAHCIFamily	288
com.apple.driver.AppleIntelLpssGspi	3.0.60
com.apple.driver.usb.AppleUSBXHCIPCI	1.2
com.apple.driver.usb.AppleUSBXHCI	1.2
com.apple.driver.usb.AppleUSBHostPacketFilter	1.0
com.apple.driver.AppleEFINVRAM	2.1
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.driver.AppleEFIRuntime	2.1
com.apple.driver.AppleBacklightExpert	1.1.0
com.apple.iokit.IONDRVSupport	519.20
com.apple.iokit.IOGraphicsFamily	519.20
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.driver.DiskImages	480.60.1
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleEffaceableStorage	1.0
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleUSBTDM	439.70.3
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.iokit.IOUSBMassStorageDriver	140.70.2
com.apple.iokit.IOSCSIBlockCommandsDevice	404.30.2
com.apple.iokit.IOSCSIArchitectureModelFamily	404.30.2
com.apple.iokit.IOStorageFamily	2.1
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.2
com.apple.driver.usb.AppleUSBCommon	1.0
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.AppleSEPManager	1.0.1
com.apple.driver.IOSlaveProcessor	1
com.apple.iokit.IOReportFamily	31
com.apple.iokit.IOTimeSyncFamily	680.15
com.apple.iokit.IONetworkingFamily	3.4
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
com.apple.kec.Libm	1

EOF
Model: MacBookPro14,1, BootROM MBP141.0178.B00, 2 processors, Intel Core i5, 2.71 GHz, 12 GB, SMC 2.43f6
Graphics: Intel HD Graphics 620, Intel HD Graphics 620, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR4, 2400 MHz, Micron, 8ATF51264HZ-2G3B1
Memory Module: BANK 0/DIMM2, 8 GB, DDR4, 2400 MHz, Kingston, 9905624-025.A00G
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x623), Broadcom BCM43xx 1.0 (7.21.190.18.1a3)
Bluetooth: Version 6.0.7f10, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: TOSHIBA MQ01ABF050, 500.11 GB
Serial ATA Device: Samsung SSD 860 EVO M.2 250GB, 250.06 GB
USB Device: USB 3.0 Bus
USB Device: BCM20702A0
USB Device: USB2.0 HD UVC WebCam
USB Device: Dell Wireless Mouse WM314
Thunderbolt Bus: 
```

Can you show me how to get more debugging information?

RehabMan, 2018-09-27:
Use the forums for troubleshooting.

RehabMan, 2018-09-28:
This should be fixed in 1.9.1 (it was reported by another user on the forums with full PR files).

closed 2018-09-28 by RehabMan

FallenChromium, 2018-10-09:
Hello. I'm facing the same problem on Acer Swift 3, absolutely same panic as above. All seems to be fine if I'm disabing VoodooI2CHID, so bug is same.

RehabMan, 2018-10-12:
If your KP is "absolutely same panic", then you clearly have an old version installed.
Latest version is 1.9.2, not 1.9.

Use the forums for troubleshooting.

ahossny, 2019-02-11:
Hi @RehabMan,
I have the same Kernel Panic when VoodooHIDI2C is loaded, I get the following Kernel panic.
I am on the latest version 1.9.2.

`Anonymous UUID:       59672747-A560-74D8-93B7-35C10DACBE6F

Sun Feb 10 14:22:56 2019

*** Panic Report ***
panic(cpu 2 caller 0xffffff800deda29d): Kernel trap at 0xffffff7f90fee567, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000022942000, CR4: 0x00000000003626e0
RAX: 0xffffff800e60d9c0, RBX: 0xffffff803d5b2ed0, RCX: 0xffffff803e8c4580, RDX: 0xffffff803ebff000
RSP: 0xffffff820a933eb0, RBP: 0xffffff820a933eb0, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0xffffff803ed5cec8, R9:  0xffffff800e69d790, R10: 0xffffff800e69da10, R11: 0x0000000000000000
R12: 0xffffff803e8c45d0, R13: 0xffffff803e8c45d8, R14: 0xffffff803ebff000, R15: 0xffffff803e8c4580
RFL: 0x0000000000010246, RIP: 0xffffff7f90fee567, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 1

Backtrace (CPU 2), Frame : Return Address
0xffffff820a933980 : 0xffffff800ddaeb0d 
0xffffff820a9339d0 : 0xffffff800dee8653 
0xffffff820a933a10 : 0xffffff800deda07a 
0xffffff820a933a80 : 0xffffff800dd5bca0 
0xffffff820a933aa0 : 0xffffff800ddae527 
0xffffff820a933bc0 : 0xffffff800ddae373 
0xffffff820a933c30 : 0xffffff800deda29d 
0xffffff820a933da0 : 0xffffff800dd5bca0 
0xffffff820a933dc0 : 0xffffff7f90fee567 
0xffffff820a933eb0 : 0xffffff800e428e3a 
0xffffff820a933f00 : 0xffffff800e42bbfb 
0xffffff820a933f50 : 0xffffff800e42d786 
0xffffff820a933fa0 : 0xffffff800dd5b0ce 
      Kernel Extensions in backtrace:
         org.rehabman.voodoo.driver.PS2Controller(1.9.2)[2FBB7D01-C031-3757-9A8A-70EF17655333]@0xffffff7f90feb000->0xffffff7f90ff6fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[67079254-1B34-31D2-8C7B-B138E26BFDD8]@0xffffff7f8ec7e000

BSD process name corresponding to current thread: kernel_task
Boot args: kext-dev-mode=1 -v -f 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 18.2.0: Thu Dec 20 20:46:53 PST 2018; root:xnu-4903.241.1~1/RELEASE_X86_64
Kernel UUID: 1970B070-E53F-3178-83F3-1B95FA340695
Kernel slide:     0x000000000da00000
Kernel text base: 0xffffff800dc00000
__HIB  text base: 0xffffff800db00000
System model name: MacBookPro15,1 (Mac-937A206F2EE63C01)

System uptime in nanoseconds: 2767907172
last loaded kext at 1862463336: com.apple.driver.AppleTopCaseHIDEventDriver	138.2 (addr 0xffffff7f905cb000, size 61440)
loaded kexts:
com.alexandred.VoodooI2CHID	1
com.insanelymac.RealtekRTL8111	2.2.1
org.rehabman.voodoo.driver.PS2Controller	1.9.2
org.rehabman.driver.AppleSmartBatteryManager	1.81.2
as.lvs1974.HibernationFixup	1.2.3
as.vit9696.AppleALC	1.3.5
as.vit9696.WhateverGreen	1.2.1
as.vit9696.Lilu	1.2.7
org.netkas.driver.FakeSMC	1800
com.alexandred.VoodooI2C	2.1.4
org.coolstar.VoodooGPIO	1.1
com.alexandred.VoodooI2CServices	1
com.apple.driver.AppleTopCaseHIDEventDriver	138.2
com.apple.driver.AppleFileSystemDriver	3.0.1
com.apple.driver.AppleVirtIO	2.1.3
com.apple.filesystems.hfs.kext	407.200.4
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.BootCache	40
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0
com.apple.AppleSystemPolicy	1.0
com.apple.filesystems.apfs	945.241.4
com.apple.driver.AppleAHCIPort	329.200.2
com.apple.private.KextAudit	1.0
com.apple.driver.AppleACPIButtons	6.1
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleACPIEC	6.1
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	190
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleHSBluetoothDriver	138.2
com.apple.driver.IOBluetoothHIDDriver	6.0.10f1
com.apple.iokit.IOBluetoothFamily	6.0.10f1
com.apple.driver.AppleMultitouchDriver	2410.5
com.apple.driver.AppleInputDeviceSupport	2410.2
com.apple.iokit.IOSerialFamily	11
com.apple.filesystems.hfs.encodings.kext	1
com.apple.driver.AppleXsanScheme	3
com.apple.iokit.IOGraphicsFamily	530.14
com.apple.iokit.IONVMeFamily	2.1.0
com.apple.iokit.IOAHCIBlockStorage	301.200.2
com.apple.iokit.IOAHCIFamily	288
com.apple.driver.usb.AppleUSBXHCIPCI	1.2
com.apple.driver.usb.AppleUSBXHCI	1.2
com.apple.driver.usb.AppleUSBHostPacketFilter	1.0
com.apple.iokit.IOUSBFamily	900.4.2
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.1
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.quarantine	3
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.DiskImages	493.0.0
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleEffaceableStorage	1.0
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleUSBTDM	456.230.1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.kext.CoreTrust	1
com.apple.iokit.IOUSBMassStorageDriver	145.200.2
com.apple.iokit.IOSCSIBlockCommandsDevice	408.200.1
com.apple.iokit.IOSCSIArchitectureModelFamily	408.200.1
com.apple.iokit.IOStorageFamily	2.1
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.2
com.apple.driver.usb.AppleUSBCommon	1.0
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.AppleSEPManager	1.0.1
com.apple.driver.IOSlaveProcessor	1
com.apple.iokit.IOReportFamily	47
com.apple.iokit.IOTimeSyncFamily	700.7
com.apple.iokit.IONetworkingFamily	3.4
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.driver.AppleACPIPlatform	6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	1.0

EOF
`

FallenChromium, 2019-02-11:
Are you sure you guys applied KextsToPatch Applelpcss patch from coolstar? Anyway, these problems can often be solved with booting in safe mode (-x) and doing 
```
sudo kextcache -i /
```

ahossny, 2019-02-17:
HI,

I was able to fix the problem.

further debugging led me to comment the following code:

`bool ApplePS2Controller::notificationHandler(void * refCon, IOService * newService, IONotifier * notifier)
{
    //_cmdGate->runAction(OSMemberFunctionCast(IOCommandGate::Action, this, &ApplePS2Controller::notificationHandlerGated), newService, notifier);
    return true;
}`

VoodooPS2Controller.cpp (Line: 2059)

I don't know the necessity of applying the following commented code, but commenting it did fix the kernel panic problem while having VoodooI2C + VoodooI2CHID installed.

@RehabMan let me know if it is necessary to apply the following code, so I can put more effort to fix the issue.

RehabMan, 2019-04-17:
The thing to test is to see why the _cmdGate->runAction causes a problem.
Maybe _cmdGate is NULL (it should not be NULL).
You could put a test in prior to the call: if (_cmdGate) _cmdGate->runAction(...)
Then investigate further why _cmdGate is NULL in the particular scenario you have.

Strange also that you have only VoodooPS2Controller loading.  Usually, you have VoodooPS2Controller plus one or more of VoodooPS2Mouse, VoodooPS2Keyboard, VoodooPS2Trackpad.  Maybe you have no PS2 keyboard or trackpad (in which case, why did you install VoodooPS2Controller?).

ErrorErrorError, 2019-09-04:
> The thing to test is to see why the _cmdGate->runAction causes a problem.
> Maybe _cmdGate is NULL (it should not be NULL).
> You could put a test in prior to the call: if (_cmdGate) _cmdGate->runAction(...)
> Then investigate further why _cmdGate is NULL in the particular scenario you have.

I unfortunately have this issue too. I cannot get the panic log from -verbose but it's the same as @ahossny 's panic log. I actually do get _cmdGate as null but I don't know why since I need it for the VoodooPS2 keyboard. This only happens when I am loading VoodooI2C for trackpad support.