RehabMan/OS-X-Voodoo-PS2-Controller #175: Panic when unloading (sleep initiated)

ESProcessing, 2019-01-01:
Version 1.8.x works fine, sleep is perfect. Update to any 1.9.x and initiating sleep generates a KP when unloading the Trackpad kext. 

This is on a Lenovo T470 (I5-6500u, HD520 1920x1024)

Anonymous UUID:       EC038489-3092-0406-30E4-8B3B5720E3DD

Tue Jan  1 14:50:59 2019

*** Panic Report ***
panic(cpu 1 caller 0xffffff800eeda1ed): Kernel trap at 0xffffff7f8f8e42cf, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000000113af8000, CR3: 0x00000003e0ec511f, CR4: 0x00000000003626e0
RAX: 0x3f00115371e9e300, RBX: 0xffffff8032aba000, RCX: 0xffffff800f69deb8, RDX: 0x0000000000000000
RSP: 0xffffff9218133a90, RBP: 0xffffff9218133aa0, RSI: 0x0000000009000000, RDI: 0xffffff8032c57400
R8:  0xffffff800f69de50, R9:  0xffffff800f69de80, R10: 0xffffff800f69deb0, R11: 0x0000000000000000
R12: 0x0000000100000326, R13: 0xffffff800f4292b0, R14: 0xffffff8032ac8ea0, R15: 0x0000000100000323
RFL: 0x0000000000010286, RIP: 0xffffff7f8f8e42cf, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000113af8000, Error code: 0x0000000000000000, Fault CPU: 0x1, PL: 0, VF: 0

Backtrace (CPU 1), Frame : Return Address
0xffffff800eb4ca90 : 0xffffff800edaeafd
0xffffff800eb4cae0 : 0xffffff800eee85a3
0xffffff800eb4cb20 : 0xffffff800eed9fca
0xffffff800eb4cb90 : 0xffffff800ed5bca0
0xffffff800eb4cbb0 : 0xffffff800edae517
0xffffff800eb4ccd0 : 0xffffff800edae363
0xffffff800eb4cd40 : 0xffffff800eeda1ed
0xffffff800eb4ceb0 : 0xffffff800ed5bca0
0xffffff800eb4ced0 : 0xffffff7f8f8e42cf
0xffffff9218133aa0 : 0xffffff7f8f8dff12
0xffffff9218133ad0 : 0xffffff800f4293bd
0xffffff9218133b10 : 0xffffff800f45546e
0xffffff9218133b50 : 0xffffff800f428479
0xffffff9218133bd0 : 0xffffff800f427577
0xffffff9218133c30 : 0xffffff800f42b15d
0xffffff9218133cb0 : 0xffffff800f44e351
0xffffff9218133ce0 : 0xffffff800f44e545
0xffffff9218133d20 : 0xffffff800f48fc24
0xffffff9218133d50 : 0xffffff800ee91eb0
0xffffff9218133d80 : 0xffffff800edb42bd
0xffffff9218133dd0 : 0xffffff800ed8ebe5
0xffffff9218133e50 : 0xffffff800eda359e
0xffffff9218133ef0 : 0xffffff800eec170b
0xffffff9218133fa0 : 0xffffff800ed5c486
      Kernel Extensions in backtrace:
         org.rehabman.voodoo.driver.PS2Trackpad(1.9.2)[806551A5-5C5D-3F7F-8EC9-4DF623790818]@0xffffff7f8f8dc000->0xffffff7f8f8e9fff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[2AEFB432-C333-3CFC-955E-24BBDDDE0F5E]@0xffffff7f8f822000
            dependency: com.apple.iokit.IOReportFamily(47)[75A625B5-0DEB-343A-A723-9D72533F098C]@0xffffff7f8f8cc000
            dependency: org.rehabman.voodoo.driver.PS2Controller(1.9.2)[622026FA-B8FF-3C80-8477-629728C99236]@0xffffff7f8f8b6000

BSD process name corresponding to current thread: kextunload
Boot args: kext-dev-mode=1

Mac OS version:
18C54

Kernel version:
Darwin Kernel Version 18.2.0: Mon Nov 12 20:24:46 PST 2018; root:xnu-4903.231.4~2/RELEASE_X86_64
Kernel UUID: 56B30885-F9BA-30E8-AD1C-5D59EC243BA9
Kernel slide:     0x000000000ea00000
Kernel text base: 0xffffff800ec00000
__HIB  text base: 0xffffff800eb00000
System model name: MacBookPro13,1 (Mac-473D31EABEB93F9B)

System uptime in nanoseconds: 125482481186
last loaded kext at 92807046175: com.apple.filesystems.msdosfs	1.10 (addr 0xffffff7f900ce000, size 69632)
loaded kexts:
(none)

EOF
[debug_25989.zip](https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller/files/2719596/debug_25989.zip)

Charlyo, 2019-02-25:
Same here, on a thinkpad T460 (I5-6200u, HD520 1920x1080)

Anonymous UUID:       B2E96F9B-57DB-D02E-E3B8-C01B4BBE9FE6

Mon Feb 25 10:24:04 2019

*** Panic Report ***
panic(cpu 2 caller 0xffffff80101fea55): Kernel trap at 0xffffff7f90c5566d, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x00007fffc47ad878, CR3: 0x0000000609a1505a, CR4: 0x00000000003627e0
RAX: 0x3f0011bdeb0f670e, RBX: 0xffffff803e521000, RCX: 0x0000000009000000, RDX: 0xffffff80108919f8
RSP: 0xffffff92d8553ae0, RBP: 0xffffff92d8553af0, RSI: 0x0000000002120001, RDI: 0xffffff803e4dac80
R8:  0xffffff803e4dac80, R9:  0xffffff80108919c8, R10: 0x0000000000000001, R11: 0x0000000000000001
R12: 0x000000010000028d, R13: 0xffffff8010693fd0, R14: 0xffffff803e486b80, R15: 0x0000000100000289
RFL: 0x0000000000010282, RIP: 0xffffff7f90c5566d, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x00007fffc47ad878, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff800ff4c450 : 0xffffff80100e956c 
0xffffff800ff4c4d0 : 0xffffff80101fea55 
0xffffff800ff4c6b0 : 0xffffff801009a5a3 
0xffffff800ff4c6d0 : 0xffffff7f90c5566d 
0xffffff92d8553af0 : 0xffffff7f90c51923 
0xffffff92d8553b20 : 0xffffff8010694093 
0xffffff92d8553b60 : 0xffffff80106bacbe 
0xffffff92d8553ba0 : 0xffffff8010693488 
0xffffff92d8553c20 : 0xffffff8010692747 
0xffffff92d8553c70 : 0xffffff8010694c1d 
0xffffff92d8553d00 : 0xffffff80106b4b41 
0xffffff92d8553d30 : 0xffffff80106b4d25 
0xffffff92d8553d70 : 0xffffff80106ed9c7 
0xffffff92d8553da0 : 0xffffff80101b9d6e 
0xffffff92d8553dd0 : 0xffffff80100ee7ec 
0xffffff92d8553e30 : 0xffffff80100cafe1 
0xffffff92d8553e80 : 0xffffff80100df7a7 
0xffffff92d8553f10 : 0xffffff80101ea038 
0xffffff92d8553fb0 : 0xffffff801009adb6 
      Kernel Extensions in backtrace:
         org.rehabman.voodoo.driver.PS2Trackpad(1.9.2)[E616F856-B251-31D5-8A8F-767E322E1C24]@0xffffff7f90c4e000->0xffffff7f90c59fff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[7F020D24-38F2-3DC4-A69C-40665E78D9F4]@0xffffff7f90baa000
            dependency: com.apple.iokit.IOReportFamily(31)[B14DC3D3-7250-3DA3-BF50-C666EBEDAF4C]@0xffffff7f90c39000
            dependency: org.rehabman.voodoo.driver.PS2Controller(1.9.2)[39822203-AB99-38FA-85DB-A83BA02A60CA]@0xffffff7f90c40000

BSD process name corresponding to current thread: kextunload
Boot args: kext-dev-mode=1 

Mac OS version:
16G1815

Kernel version:
Darwin Kernel Version 16.7.0: Thu Dec 20 21:53:35 PST 2018; root:xnu-3789.73.31~1/RELEASE_X86_64
Kernel UUID: C0FBB47C-FDCC-3009-815F-8B5C4844FEBF
Kernel slide:     0x000000000fe00000
Kernel text base: 0xffffff8010000000
__HIB  text base: 0xffffff800ff00000
System model name: MacBookPro13,1 (Mac-473D31EABEB93F9B)

ESProcessing, 2019-02-28:
> Same here, on a thinkpad T460 (I5-6200u, HD520 1920x1080)
> 
> Anonymous UUID: B2E96F9B-57DB-D02E-E3B8-C01B4BBE9FE6
> 
> Mon Feb 25 10:24:04 2019
> 
> *** Panic Report ***

I just removed the remove/load of the driver from my rc scripts. It does not adress the root of the problem, but it works just fine leaving the kext loaded while sleeping.

RehabMan, 2019-04-17:
This kext is not designed to be unloaded.

closed 2019-04-17 by RehabMan