RehabMan/OS-X-Voodoo-PS2-Controller #176: What are the Thinkpad-dsl files for?

vistree, 2019-01-04:
Hi, I am new to the Hackbook world ;-) I set up a t480s with mojave. Everything is great besides the fact that I can't use this kext to scroll on my trackpad. Is this somehow enabled by the additional thinkpad-dsl files? Do I need both - or just one of them?
It seems the Trackpad is an ELAN on - so is the Trackpoint (wiht 3 physical buttons). Any idea?

BesnikRrustemi, 2019-01-04:
Assuming the fact that most of 7th gen+ laptops have precision touchpads, most of them are I2C the difference is some are from ELAN, some from Synaptics, some ALPS

For this you should go to:
https://github.com/alexandred/VoodooI2C/issues/139

Use these kexts, use VoodooPS2Controller.kext for keyboard and I2C kexts for Touchpad

BesnikRrustemi, 2019-01-04:
@vistree if I2C doesn’t support it, VoodooPS2Controller can take care of that, you will need the SSDT-Thinkpad_TrackPad.aml to have trackpoint working.

closed 2019-01-04 by vistree