RehabMan/OS-X-Voodoo-PS2-Controller #180: Scroll and swipe with 2 fingers issue

effeci, 2019-01-29:
Hello! Thanks for the great work!! I'm using the next on a Thinkpad x240, everything is working fine but scrolling is not smooth... I mean with the single scroll I can't scroll more than 1/3 of a page. Is there any value I can modify? Swipe instead doesn't work at all. Thanks for the attention

jmichealson, 2019-02-21:
I have a Dell 7490 that appears to have the same issue. Mojave. The trackpad only moves the mouse a very small amount per swipe, the buttons don't work, and the stick moue also does nothing.  HTH, what data can I provide?

RehabMan, 2019-04-17:
Please use the forums.  Impossible to tell what is going on with the lack of information you provided here.

closed 2019-04-17 by RehabMan