RehabMan/OS-X-Voodoo-PS2-Controller #182: DELL AT101W PS2: Keymap incorrect

stillline, 2019-03-14:
MacOS Mojave 10.14.3

I'm using a Dell AT101W PS2 keyboard plugged into the ps2 keyboard on my Gigabyte B75M D3H motherboard, with the latest version of the VooDoo kext downloaded from github. 

When I press Windows Key + C I get a special character "ç" instead of the copy command. 

Thinking that maybe the Option and Command keys were swapped I checked andthe modifer keys are in fact set to defaults. 

Is this something I should handle with custom mapping in the plist? 

Apologies if this issue has already been addressed elsewhere. I did some googling beforehand and didn't see this issue mentioned. 

When I use this keyboard in windows the keys are mapped correctly.

stillline, 2019-03-14:
I was reading some of your instructions on debugging. When I run the debug kext and look and system.log I don't see any entries ps2 keyboard. 

Also when I edit the info.plsit I dont see a section for remapping keys. 

I am running the kexts in the clover/kexts/other folder on the efi partition.

stillline, 2019-03-14:
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>BuildMachineOSBuild</key>
	<string>18A391</string>
	<key>CFBundleExecutable</key>
	<string>VoodooPS2Controller</string>
	<key>CFBundleGetInfoString</key>
	<string>1.9.2, Copyright Apple Computer, Inc. 2000-2003, David Elliot 2007, RehabMan 2012-2013</string>
	<key>CFBundleIdentifier</key>
	<string>org.rehabman.voodoo.driver.PS2Controller</string>
	<key>CFBundleInfoDictionaryVersion</key>
	<string>6.0</string>
	<key>CFBundleName</key>
	<string>Voodoo PS/2 Controller</string>
	<key>CFBundlePackageType</key>
	<string>KEXT</string>
	<key>CFBundleShortVersionString</key>
	<string>1.9.2</string>
	<key>CFBundleSignature</key>
	<string>????</string>
	<key>CFBundleSupportedPlatforms</key>
	<array>
		<string>MacOSX</string>
	</array>
	<key>CFBundleVersion</key>
	<string>1.9.2</string>
	<key>DTCompiler</key>
	<string>com.apple.compilers.llvm.clang.1_0</string>
	<key>DTPlatformBuild</key>
	<string>9F2000</string>
	<key>DTPlatformVersion</key>
	<string>GM</string>
	<key>DTSDKBuild</key>
	<string>12D75</string>
	<key>DTSDKName</key>
	<string>macosx10.8</string>
	<key>DTXcode</key>
	<string>0941</string>
	<key>DTXcodeBuild</key>
	<string>9F2000</string>
	<key>IOKitPersonalities</key>
	<dict>
		<key>ACPI PS/2 Nub</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>org.rehabman.voodoo.driver.PS2Controller</string>
			<key>FindMouseDelay</key>
			<integer>100</integer>
			<key>IOClass</key>
			<string>AppleACPIPS2Nub</string>
			<key>IONameMatch</key>
			<array>
				<string>PNP0303</string>
				<string>PNP030B</string>
				<string>PNP0320</string>
			</array>
			<key>IOProviderClass</key>
			<string>IOACPIPlatformDevice</string>
			<key>MouseNameMatch</key>
			<array>
				<string>PNP0F03</string>
				<string>PNP0F0B</string>
				<string>PNP0F0E</string>
				<string>PNP0F13</string>
			</array>
		</dict>
		<key>ApplePS2Controller</key>
		<dict>
			<key>CFBundleIdentifier</key>
			<string>org.rehabman.voodoo.driver.PS2Controller</string>
			<key>IOClass</key>
			<string>ApplePS2Controller</string>
			<key>IONameMatch</key>
			<string>ps2controller</string>
			<key>IOProviderClass</key>
			<string>IOPlatformDevice</string>
			<key>Platform Profile</key>
			<dict>
				<key>Default</key>
				<dict>
					<key>MouseWakeFirst</key>
					<false/>
					<key>WakeDelay</key>
					<integer>10</integer>
				</dict>
				<key>HPQOEM</key>
				<dict>
					<key>1411</key>
					<string>ProBook</string>
					<key>1619</key>
					<string>ProBook</string>
					<key>161C</key>
					<string>ProBook</string>
					<key>164F</key>
					<string>ProBook</string>
					<key>167C</key>
					<string>ProBook</string>
					<key>167E</key>
					<string>ProBook</string>
					<key>1680</key>
					<string>ProBook</string>
					<key>179B</key>
					<string>ProBook</string>
					<key>179C</key>
					<string>ProBook</string>
					<key>17A9</key>
					<string>ProBook</string>
					<key>17F0</key>
					<string>ProBook</string>
					<key>17F3</key>
					<string>ProBook</string>
					<key>17F6</key>
					<string>ProBook</string>
					<key>1942</key>
					<string>ProBook</string>
					<key>1949</key>
					<string>ProBook</string>
					<key>198F</key>
					<string>ProBook</string>
					<key>ProBook</key>
					<dict>
						<key>WakeDelay</key>
						<integer>0</integer>
					</dict>
					<key>ProBook-102</key>
					<string>ProBook</string>
					<key>ProBook-87</key>
					<string>ProBook</string>
				</dict>
			</dict>
			<key>RM,deliverNotifications</key>
			<true/>
		</dict>
	</dict>
	<key>OSBundleCompatibleVersion</key>
	<string>1.9.2</string>
	<key>OSBundleLibraries</key>
	<dict>
		<key>com.apple.iokit.IOACPIFamily</key>
		<string>1.0.0d1</string>
		<key>com.apple.kpi.bsd</key>
		<string>8.0.0</string>
		<key>com.apple.kpi.iokit</key>
		<string>8.0.0</string>
		<key>com.apple.kpi.libkern</key>
		<string>8.0.0</string>
		<key>com.apple.kpi.mach</key>
		<string>8.0.0</string>
		<key>com.apple.kpi.unsupported</key>
		<string>8.0.0</string>
	</dict>
	<key>OSBundleRequired</key>
	<string>Console</string>
	<key>Source Code</key>
	<string>https://github.com/RehabMan/OS-X-Voodoo-PS2-Controller</string>
</dict>
</plist>

RehabMan, 2019-04-17:
You can customize the PS2 map with ACPI. Refer to *.dsl in the repo for examples.
This kext is primarily for laptops.
For desktops, typically people use USB keyboards.

wyatt8740, 2019-06-08:
You typically use USB keyboards, maybe.
Lots of people who like mechanical keyboards use PS/2 because of the abundance of PS/2 devices out there.
Thanks for the advice, though.

stillline, 2019-06-08:
@RehabMan Thanks for the reply. Turns out the simplest solution was to switch to a ps2 to USB adapter event though I have a ps2 port on this machine. Now all the keys map correctly and I don't have to use the kext at all.

closed 2019-06-08 by RehabMan

wyatt8740, 2019-06-10:
> @RehabMan Thanks for the reply. Turns out the simplest solution was to switch to a ps2 to USB adapter event though I have a ps2 port on this machine. Now all the keys map correctly and I don't have to use the kext at all.

This isn't really a reason to close this issue. There are still lots of us that have PS/2 ports, and these USB adapters don't typically translate unusual keys' scan codes. I have two keyboards with multimedia keys that aren't translated by either of my PS/2-to-USB active adapters, one of which is using an original IBM controller board (i.e., the controller uses IBM's own reference PS/2 implementation). Pressing these keys while using a USB adapter results in the adapter completely ignoring the event and not passing it through.

KeithMarcusXIII, 2021-03-02:
> I was reading some of your instructions on debugging. When I run the debug kext and look at system.log I don't see any entries for ps2 keyboard.
> 
> Also when I edit the info.plsit I dont see a section for remapping keys.
> 
> I am running the kexts in the clover/kexts/other folder on the efi partition.

@RehabMan I have the same issue with no such logs in console. I was running Big Sur 11.2

stillline, 2021-03-02:
Just use a ps2 to usb adapter.

KeithMarcusXIII, 2021-03-02:
> Just use a ps2 to usb adapter.
> […](#)

It's a laptop keyboard though......