RehabMan/OS-X-Voodoo-PS2-Controller #185: Scroll speed and right click tapping.

mallgrab, 2019-05-17:
Hey! The kext works pretty alright but there are some things that I want to tweak.

Scroll speed seems to be really slow, tweaking the touchpad settings doesn't make it fast enough even on max settings. 

Also is it possible to tweak the time where tapping two fingers creates a right click? It seems extremely strict where even a small fracture of a couple of miliseconds will not create a right click.

mallgrab, 2019-05-18:
Tweaking ScrollResolution did it for me.
Still, is there no way to configure the threshold of tapping with two fingers to right click?
It's as if both fingers have to tap at the exact moment. If one finger lands just a few milliseconds too late the right click never happens.

mallgrab, 2019-05-28:
Tweaking ScrollResolution fixed the slow scroll speed.

Timing is still an issue, might have to investigate the other forks..

RehabMan, 2019-06-08:
Refer to example *.dsl in the repo for configuration items you might wish to change, especially if you have Thinkpad.

closed 2019-06-08 by RehabMan