RehabMan/OS-X-Voodoo-PS2-Controller #186: Scrolling is choppy/rough

dreamwhite, 2019-05-27:
Hi there, I've installed Mac OS Mojave 10.14.5 using the Vanilla method. Actually it all works (except for the imessage and facetime apps but who cares xD). What actually is not working well is the scrolling while using the trackpad. I've got a Dell Inspiron 5370 and I'm using the latest build of VoodooPS2Controller kext. If I scroll a little bit (such as 1 inch) the page goes down a lot. I don't know how to explain this issue as well...

MSFT0001:00 06CB:7E7E Touchpad <= This is my touchpad model

From "xinput list" I can certify that my trackpad is a PS2 Synaptics Model. I've tried many kext such as VoodooI2C but I can't use the touchpad as well as I do on linux. Any advice?
Many thanks

mallgrab, 2019-05-28:
Tweaking the plist within VoodooPS2Trackpad.kext can improve the scrolling.
There are also several forks which have their own tweaks that could help your issue aswell.

RehabMan, 2019-06-08:
You will need to explain better, not to mention provide details on your installation.  Best to use the forums.

closed 2019-06-08 by RehabMan