RehabMan/OS-X-Voodoo-PS2-Controller #189: Swapped Caps-Lock and Control Causes Locked Control

seyhaheng2, 2019-07-01:
verything is working well finally on my HP Envy with High Sierra except for one thing...

I always use the caps-lock key as my control key. When I swap caps lock and control using either the Apple keyboard control panel or Karabiner Elements, when I press the caps-lock key (which is now control) it lights up (it has an LED in it) and at that point it locks control on so that every key I press is control modified. And it will not turn off. If I hit the caps-lock key again it does not turn off. LED stays lit and keys are all still control modified.

I have a bad feeling the keyboard controller is managing caps lock and so the OS has no control over this locking behavior.

frol, 2019-09-11:
@seyhaheng2 I have fixed a workaround for this issue in https://github.com/acidanthera/bugtracker/issues/435. Please, give it a try.

Mjoycarry, 2019-09-21:
> @seyhaheng2 I have fixed a workaround for this issue in [acidanthera/bugtracker#435](https://github.com/acidanthera/bugtracker/issues/435). Please, give it a try.

Amazing! It's so helpful for me. Thank you very much.