RehabMan/OS-X-Voodoo-PS2-Controller #193: It conflicts with voodooi2c？

qq1014853731, 2020-03-08:
It conflicts with voodooi2c. When I add voodooi2c.kext to improve my gesture operation, the startup will report an error. This is the log link [https://github.com/qq1014853731/apple/blob/master/boot.log](https://github.com/qq1014853731/apple/blob/master/boot.log)

qq1014853731, 2020-03-08:
CPU: Coffee Lake i7-9700

this is my computer info(Chinese)
https://github.com/qq1014853731/apple/blob/master/Report.htm

cu6apum, 2020-09-13:
Same here. i3 10gen, acer a515-54g. When voodooi2c enabled, the voodoops2 works ONLY when i set 'debug kexts' in clover, otherwise the kb is dead.
Updating to the latest kexts led to ps2 not working sometimes even with this option.
Any config suggestions please? 
Thank you!